mriscoc/Ender3V2S1 #2: Pro UI as a library?

thinkyhead, 2022-01-23:
I understand that LGPL mixed with a GPL project is tricky.

Here is a good answer to the question of how to mix LGPL with a GPL project and what you need to do to be able to keep your source code private while releasing binaries…

https://softwareengineering.stackexchange.com/a/342158

Basically, if you compile your sources into a library (i.e., a file ending with `.a` or `.s`) that can be linked into the larger project, then the LGPL part of the source code may be kept secret. You still have to share the modified GPL sources (most of Marlin), but you only need to include the built library and add the necessary linker parameters so that the whole thing can be built.

If you want to use code with the LGPL license within a GPL project, that is fine, but the whole of it is still GPL.

It is possible to build parts of Marlin as static-linkable libraries by placing the code into a folder called `lib` at the root of the project, with multiple files in a sub-folder. I have done this with single source files, producing an intermediate library that was then linked into the final project, and it was surprisingly easy.

Today you are in compliance with GPL by including your sources, but the re-licensing of _the whole_ as LGPL is _verboten_. So give the library trick a try and see if you can get it to work with the Pro UI code, and then you'll be able to retain 100% compliance with GPL while keeping your LGPL code secret.

Thank you for all the work you do.

mriscoc, 2022-01-23:
Thank you for your advice, I do a clarification in the front page of the repository and stablished that Marlin is GPL. But there are also many other files that I want to share under the LGPL.
I have to explore the possibility of the linked libraries. C++ is not my primary work so some tricks escape from me.

thinkyhead, 2022-01-26:
Of course all this technical fun is only required if you **must** distribute binaries while keeping the source code concealed.

The main concern for me is to stop receiving complaints on Discord from certain angry user(s) giving me grief and demanding that I take punitive actions. They are not wrong, since they are entitled under GPL to all the source code used to make a given Marlin binary — but they might be "wrong" in the extreme degree that they are angry and trying to force the situation.

Anything you can do to help restore my peaceful existence is much appreciated!

thinkyhead, 2022-01-26:
Ironic, but true.

feelfree69, 2022-02-20:
> No matter what is done, there will always be annoying users, especially if they believe they are in competition and it affects their interests.

Yes, really ironic, that a developer like you who benefits from open source projects, seems to be the only one who believes to be in a competition and tries to protect his work in closed source instead of sharing it to the community. You are stealing and try to make money with your stolen code, shame on you!

closed 2023-09-11 by mriscoc