mriscoc/Ender3V2S1 #3: SKR MINI E3 V3.0

V1ntex, 2022-01-23:
Hello!

Can anyone help correct the config files for SKR E3 Mini V3.0

I'm trying to build but I'm getting an error. 
[image]

mriscoc, 2022-01-23:
Sorry but I don't have a configuration for the SKR E3 Mini V3.0 yet, and also, I don't have that hardware to do test.

disco32r, 2022-02-16:
@mriscoc, I'm working on compiling FW for my SKR min E3 V2. If I wanted to replicate the selectable grid size feature, what directories or files would I need to copy? Or is it not that simple?

mriscoc, 2022-02-16:
The best thing that you can do is start with one of the configuration files already provided and make your changes there. Currently only the 4.2.2 and 4.2.7 boards of Ender3V2 and the 2.4.S1_301 of Ender3S1 are supported.