mriscoc/Ender3V2S1 #9: Is this firmware dead now for public use ?

enly1, 2022-01-30:
I've seen lots of messages around about this firmware and not continuing publically, but wanted to check the formal status.
Is this firmware dead now for public use ?

mriscoc, 2022-01-31:
It's not dead, check the commits. but a lot of development is in private.

josederas, 2022-01-31:
@mriscoc I believe you are bringing all your changes into the main Marlin train, yes?  I believe I saw the PROUI added to the 2.0bug-fix?  Is it assumed then that this is a specific compiled version of the main  tree?

tom-wetfrog, 2022-01-31:
Hello, can we gete a compiled .bin file, please? Need it for V2.2.7
I have no Idea how to do this

Cal-LF, 2022-01-31:
@tom-wetfrog Try this. It is my compile of the current release. V427 board.
[MRC 20220126.zip](https://github.com/mriscoc/Ender3v2/files/7974574/MRC.20220126.zip)

gokhanselen, 2022-02-01:
Yeni sürümleri nasıl indiricez yardımcı olurmusunuz

Cal-LF, 2022-02-01:
@gokhanselen There are no precompiled files to download at this time. See #6.

jeroenvermeulen, 2022-02-02:
Build instructions: #12