mriscoc/Ender3V2S1 #17: Ender3S1 - How ABL works? Probe offset / Probe margin / Build volume

mrarzt, 2022-02-03:
Hello,

thank you for the great work. I´m getting not consistent results with my first layer. In some corners they are too far away and in other to near if i check the print results.

I´m wondering because in the firmware for Ender3S1 it is default set to 230x230 build volume. So is this by intention? Did i need to change the Bed shape in my slicer to 230x230 too?

Yes or? If not the ABL points don`t match the printing points or? The thing is i can print in the corner back right but the CR touch can`t get there to probe.

So i think i need to do manual leveling or? But if i do manual leveling and then i run the Job with the ABL it says back right corner has this value which will be checked from the ABL but it is not the same point which i print or?

So the question is - how can i get sure that the ABL mesh will be put exactly on the bed size where the nozzle is????

Hopefully i have described my issue that someone can understand it. I don`t get it how i get the first layer great.

mriscoc, 2022-02-03:
The current bed size of a stock Ender 3 V2/S1 is 235x235 mm². Because the clips and offsets the usable print area is 230x230 mm² and the vendor sets the print size to 220x220 mm². For use all the printable area you must to modify the physical settings and the slicer settings.
Because the probe offsets some places in the bed are not reachable, that can generate some inaccuracies near to the border if the bed is too warped. Currently the probing area is centered, the probe margin is applied equally to the four borders. If the probe XY offset is too big, that requires that the probe margin to be increased, losing probing area. A possible solution to this is not centered the probing area and use separate probe margins for each side, to allow to use all the reachable area.

mrarzt, 2022-02-03:
Thank you for the fast answer. Just to get sure because it is my first printer.

First i manual level each corner with a paper and adjusting the wheels and z offset till i get ob each corner the paper pulled out but not pushed in.

Then i go to Z Wizard and let the nozzle go to 0 and then i do the same with the paper and lower the z offset. Thats it? If i do it with Bed Trimming it is not working good. Slicer 230x230 and good?

mriscoc, 2022-02-03:
Bed tramming allows you to do the first corners leveling in a semi assisted way, for it to work correctly you must have the correct probe margin and offsets.

mrarzt, 2022-02-03:
Yeah. So how can i get the correct values for the S1? 🙈

mriscoc, 2022-02-03:
Bed tramming must work without changes. Check if your probe margin is 40 mm or more. Try increasing your probe margin until all four points are reachable.

mrarzt, 2022-02-03:
You mean decrease or? I have it to 40 but the CR Touch Pin is „far“ away from the wheels. If i increase it the pin moves more to the inside. If i decrease to 30 it get better but in the front it is too near at the edge. So it is not the same distance. Sure that the values correct? Here you can see with 40:
[image]
[image]
[image]