mriscoc/Ender3V2S1 #19: Autolev Levelling Sensor

RPI-PMK, 2022-02-04:
Hi,

I have an Autolev sensor from Substancia 3D and trying to get it working with your great firmware. Been using Manual Mesh for a while now and had no issues so far.

They use a modified Jyers firmware which they host on their site.

I have details of changes needed in Configuration.h and Configuration_adv.h but stuck as to how to make these changes to get it working. Can you please point me in the right direction ??

I got this from supplier. How would I implement this in firmware??

Configuration on Configuration.h
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define FIX_MOUNTED_PROBE
#define X_PROBE_OFFSET_FROM_EXTRUDER 0 // X offset: -left +right [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0 // Z offset: -below +above [the nozzle]
#define AUTO_BED_LEVELING_BILINEAR
#define Z_SAFE_HOMING
#define MULTIPLE_PROBING 2
//#define MIN_SOFTWARE_ENDSTOP_Z
#define Z_CLEARANCE_DEPLOY_PROBE 3 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES 2 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE 2 // Z Clearance between multiple probes
//#define Z_AFTER_PROBING 5 // Z position after probing is done
#define Z_PROBE_LOW_POINT -2 // Farthest distance below the trigger-point to go
before stopping

Configuration on Configuration_adv.h
#define BABYSTEP_ZPROBE_OFFSET
#define DOUBLECLICK_FOR_Z_BABYSTEPPING

I have it installed on ender 3v2 and can measure the offsets but that is as far as I have got.

@mriscoc this was originally put on another of your repositories but suddenly closed and archived today. Is this something you can help me with.

Thanks

mriscoc, 2022-02-06:
That is in my to do. You can help me by comparing your config files with the files published here and list only the relevant differences, Then, I will put those differences in a json archive. You can start with the BLTouch configuration files.

mriscoc, 2022-02-15:
Have you a link to your probe and manufacturer?