mriscoc/Ender3V2S1 #22: Issue with bed tramming

davkvi, 2022-02-04:
4.2.2 bl touch

When bed tramming is enabled it will only allow the following
left front
left back
center

On the following it does nothing, only the display changes.
right front
right back

jk7gr, 2022-02-04:
I assume you had to change the X and Y offsets of your probe. If this is the case you also need to increase the probe margin. This is currently set to 20 mm. This means that if the offset you have set is bigger than then initial one then your nozzle is pushed very far right to an unreachable area. If your margin is set to 30mm then your nozzle will not be push than much to the right. I would suggest that you change the margin to 30 and test again. Let me know how it goes..

mrarzt, 2022-02-04:
Which settings did you use jk? So probe settings and home offset?

PapsDizzy, 2022-02-04:
see here guys 
https://github.com/mriscoc/Ender3V2S1/issues/15#issuecomment-1029074533

PapsDizzy, 2022-02-04:
i have increasd the margin from 25 to 30 and now is it working

jk7gr, 2022-02-05:
> Which settings did you use jk? So probe settings and home offset?

[image]

jk7gr, 2022-02-05:
I have the Satsana Air Duct which has these settings.

jk7gr, 2022-02-05:
> > Which settings did you use jk? So probe settings and home offset?
> 
> [image]

I have the Satsana Air Duct which has these settings.

davkvi, 2022-02-05:
> I assume you had to change the X and Y offsets of your probe. If this is the case you also need to increase the probe margin. This is currently set to 20 mm. This means that if the offset you have [...]

This is the correct solution :)

closed 2022-02-05 by mriscoc