mriscoc/Ender3V2S1 #25: Missing library when compile from the sources in the release 20220202

mriscoc, 2022-02-04:
If you got an error when compiling the 20220202 like this:
[image]

Please download the missing library from the repository or compile using the latest sources from the repo:

[image]

You will get a successful compile

[image]

mriscoc, 2022-02-11:
Fixed in the last release and repository

ImUrX, 2022-04-06:
You can pin issues (a maximum of 4) so they appear at the top of the search bar in the issues tab, so you dont need to have these issues open.
[image]

If you already knew this sorry to bother lol

mriscoc, 2022-04-06:
Thank you! I don't know that.