mriscoc/Ender3V2S1 #29: Bed leveling off

davkvi, 2022-02-05:
#### About the printer

**Version:** [Ender3v2-422-BLTouch-20220202.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20220202/Ender3v2-422-BLTouch-20220202.bin)
**Settings:** [Imgur album](https://imgur.com/a/VmAirX8)
**Mods:**
- Changed [mounting plate](https://i.imgur.com/dbFfV7F.png) for hotend to the ender 3 pro
- Changed [Fanduct](https://www.thingiverse.com/thing:3721746)

[image]

The red line illustrates where the probe and hotend starts.
The line is aprox 8.5mm off. It might be related to chainging the mounting plate. That does not explain why it levels all the way in the front of the bed.

mriscoc, 2022-02-05:
Change your home offsets, verify your bed size

davkvi, 2022-02-05:
> Change your home offsets, verify your bed size

Home offsets are changed.
[image]

The bltouch probes the center off the bed.

davkvi, 2022-02-05:
[image]

This image illustrates the issue now.

The x axis can be solved but the y axis is off no matter what the home is set too. I tried -50 and +50. Did not change the result apart from where the ABL probed initially to find the z height. 

Two issues i am seeing.

Probe does not use actual center for probe point. Might be the probe offset is not respected or compensated for.

mriscoc, 2022-02-06:
What happens if you increase your probe margin?

davkvi, 2022-02-06:
> What happens if you increase your probe margin?

[image]

Increasing the probe margin moves the probing further up and further inn.

mriscoc, 2022-02-06:
Then there is a problem in the mesh points definition when Home Offsets are different to zero. I will investigate but I think that it could be a problem with the Marlin core.

mriscoc, 2022-02-06:
There is a custom g-code that could help to debug this issue, the C12 g-code allows to get the bed z-value of a given X, Y point with a probe.

mriscoc, 2022-02-06:
Would it be possible for you to test with the current bugfix version of Marlin? I will try to test but I have a different setup.

davkvi, 2022-02-07:
> Would it be possible for you to test with the current bugfix version of Marlin? I will try to test but I have a different setup.

Will see if i can get around to compiling it for upstream Marlin :)

mriscoc, 2022-02-08:
Please set home offset to 0 and move those values to X_MIN_POS and Y_MIN_POS

davkvi, 2022-02-08:
> Please set home offset to 0 and move those values to X_MIN_POS and Y_MIN_POS

I cant find the x_min_pos and y_min_pos in the menu. I can compile the firmware if needed but would like to have access to the proui then.

mriscoc, 2022-02-08:
The proui library is in the repository. https://github.com/mriscoc/Ender3V2S1/issues/25

Speedy75, 2022-02-10:
maybe i have a similar problem... with this firmware bed tramming on right front and right back does nothing.
the printer dont want move and test. manually set x / y it is working.

i downgraded to version Ender3v2-422-BLTouch-20211121. here is all working

mriscoc, 2022-02-10:
Please test with this pre-release version, do not change Home Offsets, use Advance / Physical Settings / X/Y Min pos

[Experimental-Ender3v2-422-BLTouch-20220209.zip](https://github.com/mriscoc/Ender3V2S1/files/8042815/Experimental-Ender3v2-422-BLTouch-20220209.zip)
[Experimental-Ender3S1-20220209.zip](https://github.com/mriscoc/Ender3V2S1/files/8042817/Experimental-Ender3S1-20220209.zip)

mriscoc, 2022-02-11:
Fixed in the last release

closed 2022-02-11 by mriscoc

classicrocker883, 2022-08-29:
@davkvi have you fixed the issue?

because if its like the last comment, decrease **mesh inset min**, and increase **mesh inset max**. that will maximize the area, if you need to make it smaller then about increase the **min** and decrease the **max**.

as for the probe. change the probe offsets. changing the X value moves it left or right.
changing the Y value changes it front to back.

the offset of the probe point should match the center point if you do a manual tramming.

mriscoc, 2022-08-29:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides