mriscoc/Ender3V2S1 #32: Printer Killed when trying to Automesh a 9x9 - Ender 3 Pro V2

Okywan, 2022-02-06:
Tried twice, and I'm getting an error and the mesh is aborted when trying to autobuild mesh, when mesh points are 9x9, on an ender 3 pro v2 with CR-Touch. Firmware is the Ender3v2-422-BLTouch-20220202
After about 50 points there's a message on the screen stating Printer killed and asking to shutdown the printer.

mriscoc, 2022-02-06:
Disable HS mode or increase probe Z feed rate: https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-cr-touch

Okywan, 2022-02-09:
Zprobe feed rate did not work. Will test HS mode disabling.

tonysprenk, 2022-02-16:
I'm seeing the same issue. Disabling HS is a work around that seems to work.

mriscoc, 2022-02-22:
https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-cr-touch

closed 2022-02-22 by mriscoc