mriscoc/Ender3V2S1 #42: Remaining Time not working and Stuck at "Nozzle Heating"

nicop777, 2022-02-12:
Hi. First off, what a great work. Thank you! Just wanted to share my experience, just a minor bug. While doing a test print after flashing the firmware and mesh leveled my bed, the Remaining (time) is not working and stuck at 00:00. Also, the status is stuck at "Nozzle Heating". 
[image]

Cal-LF, 2022-02-12:
I am running a print now and do not have that issue. Using V427 manual mesh.
I did notice that the time remaining stays at 0 until after it starts printing.

[image]

nicop777, 2022-02-12:
Hi! I wonder why mine is bugging. May I ask if the firmware that you are using si the Feb 09 release (20220209)? Thanks!

mriscoc, 2022-02-12:
If you are using Octoprint or another host to print you must configure the appropriate plugin.

brunohenriquy, 2022-02-12:
About the "Nozzle Heating", it also happens with the Stock Marlin(bugfix-2.0.x), But there it's "E1 Heating".
It's the line 3670 from [temperature.cpp](https://github.com/mriscoc/Ender3V2S1/blob/Ender3V2S1-Released/Marlin/src/module/temperature.cpp) I Was looking the code for this Bug, then I saw that a long time ago @mriscoc made a commit there changing E to E1, then I got here and @nicop777 found the same bug as me hehehehe.

mriscoc, 2022-02-12:
"Nozzle Heating" is not a bug, it is the last message sent to the status line from the Marlin core, for change that text, you can use any M117 plugin for your slicer. For example, I use a plugin for Cura that shows the current layer number.

brunohenriquy, 2022-02-12:
> "Nozzle Heating" is not a bug, it is the last message sent to the status line from the Marlin core, for change that text, you can use any M117 plugin for your slicer. For example, I use a plugin for Cura that shows the current layer number.

Ohh Gotcha, so it sticks the last message sent, now it makes sense. Thank you @mriscoc

brunohenriquy, 2022-02-12:
> "Nozzle Heating" is not a bug, it is the last message sent to the status line from the Marlin core, for change that text, you can use any M117 plugin for your slicer. For example, I use a plugin for Cura that shows the current layer number.

Could you tell this plugin name? Link?

mriscoc, 2022-02-12:
This
[image]

brunohenriquy, 2022-02-12:
> This [image]

Thank you so much!! xD

mriscoc, 2022-02-12:
You are welcome

closed 2022-02-12 by mriscoc