mriscoc/Ender3V2S1 #43: Issue with bed tramming

rm227, 2022-02-12:
About the printer

Version: [Ender3v2-422-BLTouch-20220202.bin](https://github-com.translate.goog/mriscoc/Ender3V2S1/releases/download/20220202/Ender3v2-422-BLTouch-20220202.bin?_x_tr_sl=auto&_x_tr_tl=fr&_x_tr_hl=fr&_x_tr_pto=op,wapp)

an issue in the bed tramming menü. front right and back right are not functionable. i mean nothing happens when you chose it.

I increased the margin from 25 to 30, 35, 40,45,50 and it still doesn't work...

however before the update everything was working perfectly, I had no problem

Could you explain in detail, what should be changed to correct this problem please?

And would it be possible to give me a link to download the old version "Ender3v2-422-BLTouch-20211121." because I can't find it on github anymore, and with this version I had no problem...

mriscoc, 2022-02-12:
Hi, do you have negative home offsets? Please, post all your offsets and dimensional settings.

rm227, 2022-02-13:
Hi, thank you for your answer, here are the dimensional settings of my printer, do I need more information?

[image]
[image]
[image]
[image]

mriscoc, 2022-02-13:
About this:
>"the auto mesh seems to work, but when he finishes doing the automatic mesh of the whole tray before launching an impression, the pin of the bltouch remains deployed and rubs on the tray when moving and risks bending or breaking if it comes into contact with an obstacle"

When the auto mesh finishes the pin of the BLTouch must be stowed, if not, there is a problem of communication with your BLTouch. Please try disabling the HS mode.

rm227, 2022-02-13:
I deactivated the HS mode, now during the automatic mesh when it arrives on the right the pin does not come out and it starts all the mesh again and repeats this without stopping.

tomorrow I will disassemble the bltouch and check all the wiring to see if there is not a bad contact somewhere

mriscoc, 2022-02-13:
Don't forget to do a load defaults and store settings to clear the EEPROM with the defaults values.

Speedy75, 2022-02-14:
i have the same problem. with Ender3v2-422-BLTouch-2021112 it is working, with the latest firmware not.

mriscoc, 2022-02-14:
> i have the same problem. with Ender3v2-422-BLTouch-2021112 it is working, with the latest firmware not.

If you already do all the proposed tests then your BLTouch is not compatible with this version.

rm227, 2022-02-14:
Hi, I checked the wiring and I don't see any problem there.

I also do the "Restore defaults" and "Store settings" but the problem persists

my BLtouch is a V3.1, I don't know if that's a compatibility problem

Would it be possible to send us the version of your marlin
"Ender3v2-422-BLTouch-2021112" please?

like that, while waiting for a more recent version that is compatible, we can continue to print with our printer, because with the old version it worked perfectly
Unfortunately I can't find it on github anymore.

mriscoc, 2022-02-14:
I see posted here: https://www.facebook.com/groups/ender3v2firmware/posts/653009622407497/

rm227, 2022-02-14:
ok very good,
 thank you very much

thedroidkid, 2022-02-15:
I am having this issue as well I am unable to Tram back right and back left. I have an Ender 3 S1 with the stock CR Touch

mriscoc, 2022-02-15:
> I am having this issue as well I am unable to Tram back right and back left. I have an Ender 3 S1 with the stock CR Touch [image]

Did you change any value? Increase your probe margin.

mriscoc, 2022-02-15:
> I left everything Stock. Just flashed this morning and wanted to level my bed using the probe tram method.

Sorry, not having a S1 to do metering and test make me to do some assumptions, please test this version and report back, thank you.

[Ender3S1-20220215.zip](https://github.com/mriscoc/Ender3V2S1/files/8074235/Ender3S1-20220215.zip)

thedroidkid, 2022-02-15:
Will flash now thanks. If you need anything let me know Just picked up an s1 and sold my v2

closed 2022-02-15 by mriscoc