mriscoc/Ender3V2S1 #54: Stepper moves out of bounds when presenting print in end

davkvi, 2022-02-19:
Hi!

The y axis stepper moves out of bounds (makes clicking sounds when it hits the end of travel) when presenting the printed object at the end. Might have relation to home min x/y being a negative value?

mriscoc, 2022-02-19:
Hi, reduce the y max position and bed size.

davkvi, 2022-02-20:
> Hi, I suppose that you are referring to Ender3S1, reduce the y max position and bed size.

Ender3v2. Will do and report back

AntonyLeons, 2022-02-20:
This also happened to me on the 20220220 build

mriscoc, 2022-02-20:
> This also happened to me on the 20220220 build

Have you a stock printer, or has the printer a print head mod? If the latter is true, you must adjust the physical limits to match your configuration.

AntonyLeons, 2022-02-20:
Stock ender 3 v2

mriscoc, 2022-02-20:
> Ender3v2. Will do and report back

The Ender3v2 must have min X/Y settled to 0 in the stock, if you need to change the mins because your mods, you must reduce also the max position to match the real movement capacity of your printer.

AntonyLeons, 2022-02-20:
I concluded this happens when you start a print with the bltouch gcode. this happens to me on the X axis though At 0.

AntonyLeons, 2022-02-21:
think I figured it out, the bed it too low so scarpes the back preventing the switch from being pressed

closed 2022-02-23 by mriscoc