mriscoc/Ender3V2S1 #56: Menu hangs when changing filament

JensHauser, 2022-02-21:
**Describe the bug**
Clicking on change filament while running job -> menu hangs

**To Reproduce**
Steps to reproduce the behavior:
1. Installed fresh bin file 20220220 on an Ender 3 S1. Former original firmware 1.0.3c
2. No changed in filamant
3. Started a job
4. Pressed change filament
5. Menu hangs, clicking/turning wheel without impact

**Expected behavior**
If there is no filament added, the button should be grayed out or at least without function

mriscoc, 2022-02-21:
Thank you for your report. Can you please tell me if that behavior is with the beta standard version from the Ender3S1 repository, of from the Ender3V2S1?

JensHauser, 2022-02-21:
I installed this [version](https://github.com/mriscoc/Ender3V2S1/releases/download/20220220/Ender.3S1-20220220.bin).
Maybe you´re open for some thankful greetings to make all of this possible? I really appreciate your contribution to the 3D printer community :-)

mriscoc, 2022-02-22:
Thank you, I will review your report.

mriscoc, 2022-02-24:
I checked your report. The menu hangs because the printer is waiting for the nozzle to reach the minimum temperature that allows a filament change.

JensHauser, 2022-02-24:
Ah, thanks for showing this dependency.

closed 2022-02-24 by JensHauser