mriscoc/Ender3V2S1 #65: Black screen and steppers activated.

AdamWP, 2022-02-26:
I've been using the November release without any problems until today when the steppers started making loud noises like they are fighting to turn. Now I get a black screen and the steppers are powered and the sound from them intensifies. I've tried flashing the firmware with the latest version but the same thing happens during the flash. I'm not sure if it even flashes because the screen remains black. Could this be the firmware or is my board dead?

mriscoc, 2022-02-26:
Sorry but I think that there is a problem with the board, try to connect to a Pronterface to see if the PC can detect the board.

AdamWP, 2022-03-02:
No luck there. It's definitely the board. Thanks!

closed 2022-03-02 by AdamWP