mriscoc/Ender3V2S1 #74: Bed tramming wizard, adjust only back left knob

Crash82, 2022-03-02:
I just install Ender3S1-20220302 firmware and now when I run the bed tramming wizard its **only** saying i need to adjust the back left knob it never asks me for the other knob until i'm done.

https://i.imgur.com/2BLEoM2.png

mriscoc, 2022-03-02:
Seems that you need to raise that corner a little more.

Crash82, 2022-03-02:
> Seems that you need to raise that corner a little more.

I just updated to the new one you just released to fix a bug with octoprint and now my issue is gone, now is telling me different corner when is done.... but I have a new one 😅

When I Bed tramming and getting about 0.02 on all corners it suddenly gos to 0.13, 0.15 or something similar without even touching the knob.

mriscoc, 2022-03-02:
Each time that the printer Z axis is homed you will get a new set of values.

Crash82, 2022-03-02:
> Each time that the printer Z axis is homed you will get a new set of values.

Is home at the beginning, but after is without homing between each run ind the Bed tramming wizard

Crash82, 2022-03-02:
Here is a video of my issue and I did not touch the knob at any point under this video. 

https://youtu.be/7dFTWZ6PiHA

Woopah, 2022-03-02:
You can't Tram while cold, then level hot, and Tram again while still hot, and expect the same readings

Crash82, 2022-03-02:
> You can't Tram while cold, then level hot, and Tram again while still hot, and expect the same readings

Of course when I was recording it did it on a cold bed dammit... That's what I did for stress testing after getting home😅

I was doing a lot of testing earlier today to get it to level and I know it was on warm bed and i was get the same weird result, but I will record a new video show the issue.

mriscoc, 2022-03-03:
I saw your video and it is ok, remember the values are only relative and the bed tramming values do not necessarily correspond to the mesh values from the corners, the first bed tramming has as reference the z home made at the beginning, the second one has as reference the z home from the mesh when the bed is heating.

Crash82, 2022-03-03:
I made a new recording and I started recording when the level was almost perfect and at this part it had all ready been homed and i did not touch the knob, the only thing I did was press the button again to start the process once again it did not home in this process.

https://youtube.com/shorts/B783tit6luY

The octoprint bed visualizer was done on 60° the picture on the display was taken after a cooldown.

https://imgur.com/a/XYwQUxM
https://imgur.com/a/h8Y2PUw

mriscoc, 2022-03-03:
Your first video is ok, you level the corners, the motors got disabled, then when you do a tramming wizard again, does a new automatic z home, and the corners continue at level inside of the tolerance.
Read the wiki: https://github.com/mriscoc/Ender3V2S1/wiki/Bed-tramming-menu
[image]
The next two images are ok, you have a warped bed with the center lower than the corners.
The last picture, better to have the springs not too tight.

Crash82, 2022-03-04:
The last picture was just to show how much I have tried to level it... I know it's way too tight. 

I had overlooked the part where I said it did not need to be zero, It was just weird when the last firmware  it was not this hard to get to zero or close to so I found it was a bug. 

Thank you for your help we can close this issue now since is not a bug.

closed 2022-03-04 by Crash82