mriscoc/Ender3V2S1 #77: Printer beeps twice, shows warning-"Nozzle or bed temperature is too low", printing stops.

Tillerman6, 2022-03-03:
**Describe the bug**
A clear and concise description of what the bug is.
Loss of Nozzle and bed heat temperature during an extended print.
**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'  Thingiverse- 2799169  Spatula Holder  (large file size.stl file) 83.1 Mb, gcode file is 12.008Meg from Prusaslicer 2.4.0
2. Click on '....'
3. Scroll down to '....'  Gcode transfer on SD card as normal, 
4. Ender 3 V2 with BL touch and 4.2.2 mainboard- screen shows file name, runs as normal for about two hours and prints about 10 layers-  approx 110mm x 140 mm x 1mm.  .2mm layer height, 220 /55 temps, 

**Expected behavior**
A clear and concise description of what you expected to happen.
(((((((Print should finish.))))))))
**Screenshots**
If applicable, add screenshots to help explain your problem.Screen warnings last only 2 seconds- unable to get a photo.

**Version (please complete the following information):**
Firmware version and release date 2.0.9.4 MRiscoC   Build Dateline Mar 2 2022 02:04:34
 this is Maintenance 20220302  

**Additional context**
Add any other context about the problem here.  I did run a smaller file prior to this which lasted 5-6 hours without any issues- 4.2 mb in Gcode.

mriscoc, 2022-03-04:
Please, do a PID for Nozzle and Bed at the temperatures that you use, then save settings.

mriscoc, 2022-03-10:
No more feedback

closed 2022-03-10 by mriscoc