mriscoc/Ender3V2S1 #78: Bed Size from Configuration.h are not loaded to Printer Settings

oldo439, 2022-03-04:
Hi. I have a Creality Ender3v2 board 4.2.7 printer with BLTouch and I use your Source Code (current version Ender3v2-427-BLTouch-20220302), which I edit and compile because I also have a laser and a CNC milling machine on the printer. I have laser adjustments there.
And I found that if I set in Physical Limits Configuration.h as:
// The size of the printable area
#define X_BED_SIZE 230 // MRiscoC Max usable bed size - my bed size
#define Y_BED_SIZE 227 // MRiscoC Max usable bed size

I compile the source to BIN and burn the firmware to the printer, so my values are not transferred. In the Physical Settings menu I have Bed Size X = 230, Y = 230.

Also when I set:
#define X_MAX_POS 245 // MRiscoC Stock physical limit - my limits
#define Y_MAX_POS 230 // MRiscoC Stock physical limit
#define Z_MAX_POS 220 // Ender3v2 Configs - limit for dual-Z axis

so in the Menu I see: Max X = 248, Y = 231, Z = 250
Why not use those values from Configuration.h?
I also tried Restore Default, but it didn't help.

mriscoc, 2022-03-04:
You can use a g-code configuration file with custom g-codes to set the physical dimensions: https://github.com/mriscoc/Ender3V2S1/wiki/Custom-C-gcodes

oldo439, 2022-03-04:
OK. I just wanted to know where the problem is. Whether it's a bug in the firmware or it just doesn't load from configuration.h during compilation. I change the firmware quite often and every time I have to manually adjust the Dimensions in the Menu. G-code is usable, but not practical.
I assume the problem will be in the original Marlin code.

oldo439, 2022-03-04:
I found something in file: mriscoc\Marlin_Ender3v2\Marlin\src\lcd\e3v2\proui\dwin_defines.h
```
#if ProUI
  #undef X_BED_SIZE
  #undef Y_BED_SIZE
  #undef X_MIN_POS
  #undef Y_MIN_POS
  #undef X_MAX_POS
  #undef Y_MAX_POS
  #undef Z_MAX_POS
  #undef NOZZLE_PARK_POINT
  #if HAS_MESH
    #undef GRID_MAX_POINTS_X
    #undef GRID_MAX_POINTS_Y
    #undef GRID_MAX_POINTS
  #endif
  #if HAS_BED_PROBE
    #undef PROBING_MARGIN
    #undef Z_PROBE_FEEDRATE_SLOW
  #endif
  #undef INVERT_E0_DIR

  #define X_BED_SIZE (float)HMI_data.x_bed_size
  #define Y_BED_SIZE (float)HMI_data.y_bed_size
  #define X_MIN_POS  (float)HMI_data.x_min_pos
  #define Y_MIN_POS  (float)HMI_data.y_min_pos
  #define X_MAX_POS  (float)HMI_data.x_max_pos
  #define Y_MAX_POS  (float)HMI_data.y_max_pos
  #define Z_MAX_POS  (float)HMI_data.z_max_pos
  #define NOZZLE_PARK_POINT {(float)HMI_data.Park_point.x, (float)HMI_data.Park_point.y, (float)HMI_data.Park_point.z}
```
But I don't understand why is for ProUI this:
#undef X_BED_SIZE
and then
#define X_BED_SIZE (float) HMI_data.x_bed_size

I don't know, what is HMI_data

I before using:
#define DWIN_CREALITY_LCD_ENHANCED  // Enhanced UI
and now is there:
 #define DWIN_LCD_PROUI              // MRiscoC Enable ProUI

mriscoc, 2022-03-04:
If you disable the ProUI feature you will be able to set your physical dimensions using the configuration files but also you will lose the ability to edit the values from the menu.

oldo439, 2022-03-05:
I understand. But those default numbers must be entered somewhere in the source code. Can we find them? I would change it there if it can't be changed in Configuration.h
Even so, I adjust the settings with each new firmware. So I'll add the bed dimensions to the change list.

mriscoc, 2022-03-05:
In the file `dwin.cpp`
Adds to **DWIN_SetDataDefaults()**: 
```cpp
void DWIN_SetDataDefaults() {
  TERN_(DEBUG_DWIN, SERIAL_ECHOLNPGM("DWIN_SetDataDefaults"));
  DWIN_SetColorDefaults();
  DWINUI::SetColors(HMI_data.Text_Color, HMI_data.Background_Color, HMI_data.StatusBg_Color);
  TERN_(HAS_HOTEND,             HMI_data.HotendPidT = PREHEAT_1_TEMP_HOTEND);
  TERN_(HAS_HEATED_BED,         HMI_data.BedPidT    = PREHEAT_1_TEMP_BED);
  TERN_(HAS_HOTEND,             HMI_data.PidCycles  = 5);
  #ifdef PREVENT_COLD_EXTRUSION
    HMI_data.ExtMinT = EXTRUDE_MINTEMP;
    ApplyExtMinT();
  #endif
  #if HAS_HEATED_BED && ENABLED(PREHEAT_BEFORE_LEVELING)
    HMI_data.BedLevT = PREHEAT_1_TEMP_BED;
  #endif
  TERN_(ProUI,                  ProEx.SetDefaults());
  #if ENABLED(BAUD_RATE_GCODE)
    HMI_data.Baud115K = false;
    SetBaud250K();
  #endif
  HMI_data.x_bed_size = 230;
  HMI_data.y_bed_size = 227;
  HMI_data.x_max_pos  = 245;
  HMI_data.y_max_pos  = 230;
  HMI_data.z_max_pos  = 220;
  ProEx.ApplyPhySet();
}

oldo439, 2022-03-05:
Great, that's exactly what I needed. Thank you very much. I'll test it tomorrow.

oldo439, 2022-03-06:
It works. Thank you once again.

closed 2022-03-06 by oldo439