mriscoc/Ender3V2S1 #84: Ender3 S1 Mesh Values Wildly Deviate as Mesh Size Changes

zenforic, 2022-03-09:
**Describe the bug**
Depending on the mesh size chosen, the values at different points will be wildly different than the same or near the same points in higher or lower point meshes. Not only that, but after changing the Z-offset after an auto home (or z-home) ***OR*** doing it right after the points have been measured and mesh saved; I will move to the front part of the bed where the values deviate most and the paper I use to change z-offset is almost always heavily gripped between the nozzle and bed when it was fine at the center.

**To Reproduce**
Steps to reproduce the behavior:
1. On an Ender 3 S1 running the latest S1 specific firmware, preheat the bed to usual temps.
2. Advanced -> Mesh size -> try different sizes after repeat, starting with 4x4 (or 3x3).
3. Take photos of the meshes to check deviation over points under the same heating cycle.
4. See possible high deviation with error on mesh values at certain points (test with paper and moving x and y to different points with z at 0 and A4 paper underneath, both just after mesh and then homing).

**Expected behavior**
An auto-level z-compensation mesh with +- 0.002mm (tested with m48) values to account for bed warping on a levelled bed.

**Screenshots**
Please see below all the different mesh creations under the same thermal profile on a CR touch on the Ender 3 S1:

[image]
[image]
[image]
[image]
[image]

NOTE: The screenshots with the button over continue do not mean I chose continue, I saved each of these and did the aforementioned tests.

**Version (please complete the following information):**
Ender3 V2/S1 Professional Firmware 20220302 - maintenance release II - March 3rd, 2022

**Additional context**
Any help or insight is appreciated, other than this issue I really like this firmware! I have the Z feed rate in the probe settings set to 960 if that's any help, as per your mesh guide mentioning the CR-touch support. Could this be part of the problem?

mriscoc, 2022-03-09:
Hola, lo más interesante es la tercera captura, la impresora hace el z home (z=0) en el centro de la cama, luego se espera que el punto central de la malla esté cerca de cero, si no, entonces su cama está no en estabilidad térmica y se está deformando. Espere unos 10 minutos y tome una malla de 5x5, el punto central debe estar cerca de 0.

mriscoc, 2022-03-09:
Hi, the most interesting is the third capture, the printer does the z home (z=0) at the center of the bed, then the center point of the mesh is expected to be close to zero, if not, then your bed is not in thermal stability and it is deforming. Please wait about 10 minutes and take a 5x5 mesh, the center point must be close to 0.

zenforic, 2022-03-09:
Hello, thank you for your response. I see, that makes sense actually, I will try this and report back, thank you.

zenforic, 2022-03-10:
I've done the tests again and waited 20 minutes before starting. Again it is off on the center which is odd, will I need an enclosure for the printer to reduce thermal runaway? Is there a way to get this stable without that?

[image]

zenforic, 2022-03-10:
Decided to also do a 6x6 mesh under the same heat cycle (did not cool/reheat since the last 5x5 mesh). Here is the result of that:

[image]

Interestingly, some points close to the center to the right are quite close to 0 in this mesh.

mriscoc, 2022-03-10:
That's very informative, I think that there is a problem with how the S1 uses the CRTouch. The Ender 3V2 and 3S1 share the ABL code then it must work similarly in both printers but, I saw some S1 doesn't work fine with the current Marlin code for ABL, it is not a general failure, maybe that is the reason why the official S1 firmware only does a 4x4 mesh. I hope that someone can install a 3D/BLtouch and test if there is the same bug.

mriscoc, 2022-03-10:
I don't have a S1 for do test, but could you do a 5x5 mesh with HS disabled?

zenforic, 2022-03-10:
Yes, I'd be happy to contribute by doing that test. Do I keep the Z feed rate at 960 while having HS mode disabled?

zenforic, 2022-03-10:
I went ahead and did it with HS mode disabled, here is a 5x5 mesh with those settings:

[image]

mriscoc, 2022-03-10:
Have you access to a terminal as Octoprint or Pronterface? Decrease the Z Feed rate. I want to know if we can do a manual probe of the center point. After the Z home, that point MUST be near to 0.

zenforic, 2022-03-10:
Yes, I will try this soon.

As for having access to a terminal, I don't yet, but I had planned to set one up, and can quickly. Only problem is when for example, looking at your octoprint guide, you mention the need to cover one of the USB pins; is this necessary for the S1 and/or the use of pronterface on a PC connected to the S1 via the serial drivers instead? Oddly enough I do not have access to any electrical tape at the moment.

What gcode would you want me to run through that interface?

zenforic, 2022-03-10:
Here is a 5x5 mesh, HS mode off, Z feed rate 960, bed temperature 0:

[image]

Here is another 5x5 with everything the same except HS mode on for testing:

[image]

As an aside, I did not see an option for bed temp in advanced -> probe settings. But it did still happen at cool temperatures. Normally I manually preheat the bed before running ABL.

mriscoc, 2022-03-10:
Ok, the idea of the test is to have the heaters off when the mesh is ongoing to discard the interference of heating signals. I want to try the g-code G30 X115 Y115.

zenforic, 2022-03-11:
I have performed this action, HS Mode was still on from when I did the second test, I can turn it off again if needed. Here is trhe result from pronterface:

[image]

mriscoc, 2022-03-11:
Mmm very interesting. So, something is wrong with the bilinear calculus, but why it is working good in the V2?

zenforic, 2022-03-11:
It's a good question.. is there any difference in how it's done in the recently released S1 firmware code vs how it's implemented in the v2? If not, it's likely that that method on the S1 only works with smaller meshes as you previously mentioned for some reason.

mriscoc, 2022-03-11:
Can you do a 3x3 test? that also have a center point.

zenforic, 2022-03-11:
Here are 2 3x3 tests, first is with HS mode and the other is without HS mode:

[image]

[image]

mriscoc, 2022-03-11:
What value is your probe margin, can you set it to 50 and do a 3x3?

zenforic, 2022-03-11:
Wow, seems to make sense now since it is 0 at the center; it was at 3 which was the default. What do I need to look for to know what this value should for future reference? Also this result was in HS mode:

[image]

zenforic, 2022-03-11:
Update: I tried using the bed tramming wizard as a test as well, is it supposed to be this far into the corners? 

[image]

Also, potential other issue, is it supposed to say front right in this case? It would seem like the back right would be the one to adjust correct?

[image]

mriscoc, 2022-03-11:
The bed tramming wizard also uses the probe margin, the last picture is ok, front right is -0.02 it is the farthest value from the average.

zenforic, 2022-03-11:
I see, so is this the correct way both things are supposed to be running in general? Or was this only needed as a test for some value(s)? Because normally I'd think a mesh would be covering far corners as well like it almost was previously.

mriscoc, 2022-03-11:
In the S1, it is physically impossible for the CRTouch to reach back and right edges. Then if you want a centered mesh, you need a large probe margin, if your probe margin is short the probe tries to go to the reachable area. That is why you didn't get a 0 in the center point, because the center of the mesh was not corresponding to the center of the bed. Along of all your tests the values were consistent and not random.

zenforic, 2022-03-11:
I see, makes sense, I guess one thing was seen for this: if the default values are hard coded within the firmware, which I imagine they are, then perhaps a thing that came out of this was that for the Ender 3 S1 specific firmware, 3.0 is perhaps not a good default value. Either way, I appreciate the help, and the work you're doing here! Feel free to let me know if you'd like me to do any more tests on the S1.

mriscoc, 2022-03-11:
I first code a probe margin of ~47 but many users don't agree, then Creality releases the official sources, I took the 3 mm for the probing margin from their configuration files. But I suspect that the sources don't have the final configuration values from the lates released binaries. If you flash the official firmware, can you measure the real margin?

zenforic, 2022-03-11:
I think so, do I need to wipe my settings before flashing the official firmware?

As another potentially easier idea, if this is good enough, I looked at a review video on the section where the reviewer ran the auto-level, took a screenshot:
[image]

and then moved my axes at the same point and measured, it really is a small margin, 3mm seems about right. This is also why I was expecting it to measure corners, because it's what mine did stock. I will link the video with the timestamp in the link so it jumps straight to the ABL demonstration if you would like to see that. Otherwise just to be sure, I can re-flash the firmware again and take the measurements as it goes.

https://youtu.be/Wpk4a4ucdD4?t=262 (video source I used).

zenforic, 2022-03-11:
I also noticed just now that in that video, it also skipped the center by quite a bit, but likely because it is a 4x4 mesh.

mriscoc, 2022-03-11:
Have you Telegram? We have a group: https://t.me/ender3v2s1firmware and I will publish an updated firmware for the S1.

zenforic, 2022-03-11:
Oh yes! Sorry I forgot to join it earlier, donated on Patreon and forgot to join via the link in the email when I meant to. I will join now. Thank you!

mriscoc, 2022-03-11:
Then, time to close the issue...

closed 2022-03-11 by mriscoc