mriscoc/Ender3V2S1 #85: Mesh 7x7 to 9x9 not supported on S1

skilly00, 2022-03-09:
**Describe the bug**
Ender 3S1 does not support Mesh of 7x7 to 9x9.  When attempted a string of retries followed by the printer "killing itself"

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced - Mesh Size  
2. Click on 7x7 or 8x8 or 9x9
3. Go to Prepare - Build Mesh
4. See error - Error after creating the Mesh and retries without working

**Expected behavior**
Display of Mesh from points

**Version (please complete the following information):**
3/2/22 S1.  I"m not sure if it worked on earlier versions

Woopah, 2022-03-09:
Miguel has addressed this...you need to DISABLE HS in firmware...I just ran a 10x10, yesterday, following many fails past 6x6

mriscoc, 2022-03-09:
Please always look for similar questions:
https://github.com/mriscoc/Ender3V2S1/issues/32

Also, in the wiki :
https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-cr-touch

closed 2022-03-09 by mriscoc