mriscoc/Ender3V2S1 #87: Compiling error

araziforum, 2022-03-10:
I get some errors like that. Can anyone help to resolve that?

PS C:\Users\Selim\Desktop\Ender3V2S1> platformio run -e STM32F103RE_creality
Processing STM32F103RE_creality (board: genericSTM32F103RE; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Processing Professional Firmware requirements
Auto Mesh Bed Leveling detected
FileNotFoundError: [Errno 2] No such file or directory: 'Marlin/lib/proui/libproui_abl.a':
  File "C:\Users\Selim\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 179:
    env.SConscript(item, exports="env")
  File "C:\Users\Selim\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Selim\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Selim\Desktop\Ender3V2S1\buildroot\share\scripts\proui.py", line 24:
    shutil.copy(libpath+'libproui_abl.a', libpath+'libproui.a')
  File "C:\Users\Selim\.platformio\python3\lib\shutil.py", line 427:
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\Users\Selim\.platformio\python3\lib\shutil.py", line 264:
    with open(src, 'rb') as fsrc:
======================================================================================================================================= [FAILED] Took 1.48 seconds =======================================================================================================================================

Environment           Status    Duration
--------------------  --------  ------------
STM32F103RE_creality  FAILED    00:00:01.479
================================================================================================================================== 1 failed, 0 succeeded in 00:00:01.479 ==================================================================================================================================
PS C:\Users\Selim\Desktop\Ender3V2S1> echo "done" >"C:\Users\Selim\AppData\Local\Temp\ipc"
PS C:\Users\Selim\Desktop\Ender3V2S1>

mriscoc, 2022-03-10:
This fill fix you issue: https://github.com/mriscoc/Ender3V2S1/issues/25

araziforum, 2022-03-10:
Thx for your interest. It's done.

closed 2022-03-10 by araziforum