mriscoc/Ender3V2S1 #95: I feel like since updating to march 14 release i'm seeing issues with overextrusion?

mattfox27, 2022-03-17:
After updating to march 14 release for Ender S1 I feel like prints are possibly overextruding?  I have a part that I print and it has threads and a bolt, prior to march 14 release it fit fine now after march 14 release the parts no longer fit?  Same file.

**Additional context**
Add any other context about the problem here.

mriscoc, 2022-03-17:
Over extruding can be controlled only by the extruder steps/mm and flow. Calibrate the extruder steps/mm according to your filament and setup
[image]

nicop777, 2022-03-18:
Mine is overextruding as well. then i returned to previous version, still overextruding. Then i tried going back to jyers, it fixed the problem. then i switch back to mriscoc latest firmware, overextruding again. i wish this is fixed because overall, i liked this firmware than jyers. im using ender 3 v2 manual mesh 4.2.2 btw

mriscoc, 2022-03-19:
The extruder motor is under control of the Marlin core, if you calibrate the e-steps then the core simply follows the commands from the G-code file. We don't have reports about over extruding.

mriscoc, 2022-03-22:
> Mine is overextruding as well. then i returned to previous version, still . Then i tried going back to jyers, it fixed the problem. then i switch back to mriscoc latest firmware, overextruding again. i wish this is fixed because overall, i liked this firmware than jyers. im using ender 3 v2 manual mesh 4.2.2 btw
> 
> edit: im recalibrating e-steps every after flashing firmware

For Ender 3v2 definitely there aren't reports of over extruding for that printer. The OP refers to the Ender3S1

mattfox27, 2022-03-22:
Yes I went back to stock firmware and the problem went away, yes I also calibrated E-steps

enzu4l, 2022-03-24:
Can’t confirm that. Maybe tighten your screws. Especially the hidden screw for adjustments on the z axis gets lose after a while. Recently I had that problem and it gave me nuts.

mriscoc, 2022-04-01:
No more feedback, it can't be replicated by the test group.

closed 2022-04-01 by mriscoc