mriscoc/Ender3V2S1 #97: v20220314 Home Z is not 0 anymore

acritum, 2022-03-18:
After updating from 20220220 to 20220314 I noticed that "auto home" and "home z" do not put the nozzle to the 0 point. The nozzle is raised up about 5-10 millimeters above the table (offset shows "5"). Yes, I tried to reset to default settings but it doesn't change anything. Auto home offsets are all set to 0 in the settings. I  don't know if it was intentionally changed in the firmware for some unknown reason, but I used "home" feature  to test and fix the distance between the nozzle and the table, and now it is not possible to do it because the nozzle is always above the table when using auto home....

mriscoc, 2022-03-19:
Hi, that behavior is intentional since  z-offset persistence was added to a manual mesh version as safety measure to avoid possible nozzle crash into bed in successive movements.

acritum, 2022-03-19:
> Hi, that behavior is intentional since z-offset persistence was added to the manual mesh version as safety measure to avoid possible nozzle crash into bed in successive movements.

The funny thing is that when you select auto home  or home Z it DOES drops Z-height to 0 and only after that it raises the nozzle, so evidently it it NOT "safety measure to avoid possible nozzle crash into bed". If the nozzle reached 0 already, it won't go below that level so it can't be worse anyway, so what's the use of raising it up after it already reached the bottom?

Anyway,  what do you think is the correct / the fastest way of aligning the nozzle to the table with this version? I don't use the mesh, I usually used to set the nozzle height to Z=0 with disable steppers and used the screws under the table to set the table so that a piece of paper can go between the nozzle and the table in different parts of the table. I don't know, maybe the mesh is useful with BL-touch  but I think it's a waste of time without it because there are a lot of reasons why the alignment of the table can be changed with time (printer shifting, temperature deformations of the table, bad springs). so if one wants really good prints it's not like set the mesh and forget it till ages, one needs to check and update it quite frequently to ensure good quality of prints and it requires quite a lot of time in comparison with old classic alignment methods. I would appreciate it if you could add a new submenu item near disable steppers and homing submenus to set Z=0 and disable steppers at once with one click. Now these features are in different submenus and it is even less convenient than in the stock firmware where the menu items auto home and disable steppers go one after another.

mriscoc, 2022-03-19:
The safety measure is for movements after the home and a bad saved z-offset. For test the distance between the nozzle and the bed I suggest to use the bed tramming option in prepare menu.  Maybe it could be interesting to enable something similar to the z probe wizard but for manual mesh, in that menu, there is a Move Z to 0. Let me think about your suggestions.