mriscoc/Ender3V2S1 #99: ABL not probing any point with custom probe offsets

melvinmicallef, 2022-03-20:
**Describe the bug**
I am currently on the latest version of the firmware. Compiling my own bin file using the 422-BLTouch config files with slight modifications to the offsets results in the printer not building the ABL mesh and also, tramming to be unavailable for all 5 points (including center). I think there is an issue how the probing locations are being calculated since I have followed the formula in your WIKI to make sure that all points are probable but it still does not probe. My nozzle to probe offset is X -46, Y -6, Z -2.56 and Probe margin is set to 30. I have uploaded a video of the issue and my config files.

Please note that the same config file on version https://github.com/mriscoc/Ender3V2S1/commit/544ad9905956b205dce6eed1c822fa87acb9f8c2 still works, and that is why i am suspecting an issue with the code rather than my config file.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the config files i have provided
2. Flash the firmware
3. Try to use bed tramming or auto build mesh

**Expected behavior**
Printer should probe points areas across the bed

**Screenshots**
https://youtu.be/Jb591ZdaSPw

**Version (please complete the following information):**
https://github.com/mriscoc/Ender3V2S1/commit/5562e7535c84155f3ec359b1b9f776ef1a0499ff

**Additional context**
[Config.zip](https://github.com/mriscoc/Ender3V2S1/files/8311127/Config.zip)

mriscoc, 2022-03-24:
I'm waiting for some free time to test your configuration files.

melvinmicallef, 2022-03-29:
Any update on this please? If you require any help from my side do let me know. Maybe adding some logging to help you figure out any issues

mriscoc, 2022-03-29:
I was able to replicate your issue with the given configuration files.
Start with unaltered configuration files and do one change at a time. Identify the parameter that results in your issue.

mriscoc, 2022-04-01:
I got time to review your configuration files and noted that you disabled FILAMENT_RUNOUT_SENSOR, due the structure of the ProUI you can't disable it, if you don't have a filament run out sensor the printer will simply ignored it. Enabling that feature in your configuration files solves your issue.

melvinmicallef, 2022-04-01:
Ah was doing some testing today and came to the same conclusion. Enabling the filament sensor option does work normally. Very strange issue indeed, thank you for your help !!

closed 2022-04-01 by melvinmicallef