mriscoc/Ender3V2S1 #100: Failed flash ender 3 s1

TopazAaron, 2022-03-23:
**Describe the bug**
I just bought a ender 3 s1 and was going to update the firmware to this one as ive used this one on both my other ender v2, I ended up only getting the splash logo and would never flash. had to reddit and someone found out how to flash another firmware for the laser engraver firmware to get it working again. I'd like to use this firmware.

mriscoc, 2022-03-23:
Follow the steps in Wiki: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-install-the-firmware

closed 2022-03-23 by mriscoc

TopazAaron, 2022-03-24:
So this didn't fix my issue. I've tried and tried. Apparently mine came with a new chip stm32f4 chip which doesn't allow your firmware to flash.