mriscoc/Ender3V2S1 #107: need more firmware tweaking

HERO5507, 2022-04-01:
**Describe the bug
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date

**Additional context**
Add any other context about the problem here.

mriscoc, 2022-04-02:
Hi. There are at least 5 firmware flavors please fill in the issue template.

mriscoc, 2022-04-06:
Maybe related to the z-offset bug fixed in the latest version.

closed 2022-04-06 by mriscoc