mriscoc/Ender3V2S1 #111: CV Laser Support

Avedena, 2022-04-06:
**Is your feature request related to a problem? Please describe.**
If you want to use the Creality CV Laser its needed to flash the Board  with the Creality Firmware.

**Describe the solution you'd like**
Id like to use this Marlin Firmware with LightBurn. For this, its needed to enable the Marlin Spindle/Laser Support in Firmware.
https://marlinfw.org/docs/configuration/2.0.9/laser_spindle.html

**Describe alternatives you've considered**
For every Laser Job Flash the Firmware with the Creality Stock one.

**Additional context**
I am a newcomer to 3D printing / laser engraving. I have already tried to compile the firmware in Visual Studio Code myself, but LightBurn still does not recognize the laser.

With the Creality firmware I can move the axes, but I quickly get the error that something is wrong and I should fix the (unknown) error and restart the laser with code M999.

With the Ender3V2S1 firmware, not even the axes can be moved.

mriscoc, 2022-04-06:
Hi, thank you for your feature request. Currently, adding support for the Laser unit to Professional Firmware requires some work and time. I hope that someone can do a pull request adding this feature. I don't have an S1 and not a Laser Engraver to test, so for me it could be a little difficult to give a firmware with the tests that I expect.

mriscoc, 2022-04-06:
The Sources for the Offcial firmware with laser module are here: https://github.com/CrealityOfficial/Ender-3S1/tree/ender-3s1-lasermodel
We must port the Laser code without affecting the core Marlin code.

mriscoc, 2022-04-06:
Also you can check here: https://github.com/mriscoc/Marlin_Configurations/issues/2

Avedena, 2022-04-06:
I will digg deeper in this topic.
Can you explain, how I change the machine type in the Marlin auto build tool?
I think I was building for the wrong Machine.
[image]

mriscoc, 2022-04-06:
> I will digg deeper in this topic. Can you explain, how I change the machine type in the Marlin auto build tool? I think I was building for the wrong Machine. 

You must copy the correct configuration files, from the Configuration folder to Marlin Folder.

Avedena, 2022-04-06:
I see, but what are the diference between the xfer, maple and none?
And also the 256K and 512K?
I build the STM32F103RE_creality (512K) and sucessfully installed to the board.
Now I'am working on the port.
Thanks.

mriscoc, 2022-04-06:
For the S1 use the STM32F103RC_creality (that depends on the type of chip that your control board has)

Avedena, 2022-05-21:
> > I think the first modification is to allow Z-Homing without the BL-Toutch and insted use the Z-Linmit switch.
> 
> I don't think Z-homing is important for the laser since you need to do it manually depending upon the height of the item that you're cutting/engraving so it definitely shouldn't try to use a limit switch. When homing using the Creality laser firmware it only homes on X/Y.

You're right. At the time writing this, I don't have enough experience with it.
I was thinking the additional Z-end stop is for the Laser Head.
But I'm still working on this.

aydevosotros, 2023-01-12:
Hi there! I'd like to deep into the topic. I've been working into porting the code from the original repo but I'm getting this issue when compiling. I don't know if someone can give me a tip for it:

[image]

For what I understand it means that the code does not fit into the board memory but it doesn't make much sense for me since I didn't add much code at this point. Am I missing something?

Avedena, 2023-01-12:
Hi @aydevosotros ,
that is right. To save some space, you can disable some UI features.

mriscoc, 2023-01-12:
The STM32F4 boards have a large bootloader, so the program memory space is shorter than others boards.

TheWebMachine, 2023-01-13:
I, too, am interested in laser support. What all would have to be removed from the Ender3S1-F4-UBL firmware to make room for laser support? How thin are we on space? Do we have laser code readily able to to be combined into a build?

mriscoc, 2023-02-28:
Preliminary laser support is being discussed here: https://github.com/mriscoc/Ender3V2S1/discussions/62

closed 2023-05-11 by github-actions[bot]