mriscoc/Ender3V2S1 #115: The Setting from Autolevel in EEPROM don´t get stored.

Experte9, 2022-04-11:
### Steps to Reproduce

1. Set the hook on Autolevel
2. Press the Save button
3. Go to Terminal
4. Write M501 in terminal
5. Go back to EEPROM and u see the hook is gone

Happens also if u trun off Printer an Octoprint for more than 10 min.

[image]

daniel-volz, 2022-04-11:
In my case (i´m using a Ender 3 S1) i´m not able to Enable Autolevel using OctoPi.
The Firmware crashes and then restarts.

Experte9, 2022-04-11:
> In my case (i´m using a Ender 3 S1) i´m not able to Enable Autolevel using OctoPi. The Firmware crashes and then restarts.

Using also a S1, works fine so far did u read this? 
CAUTION! It is confirmed that Creality is selling some S1 with a STM32F4 chips, The current firmware version is incompatible with this SoC, only boards with STM32F1xx are supported. [https://github.com/mriscoc/Ender3V2S1/releases/tag/20220330](url)
Also make sure u donloaded the right .bin

daniel-volz, 2022-04-11:
Yes, Board is STM32F103, Firmware works as it should, Autoleveling is working for me, recognizable by the two blue lines under the Z symbol during printing.

Using the latest Release for ender 3 s1.

daniel-volz, 2022-04-12:
Which Bed Leveling GCode do you use?

Experte9, 2022-04-12:
I dont use any GCode for leveling, only do Leveling with the display.
Then I print, havent look for the blue Lines under Z, but i think i never saw it.
Thats my Start G-Code: 

_M104 S185 ; preheat extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
G28 ;Home
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up_

daniel-volz, 2022-04-12:
Okay, i´m using the update Leveling Button in Octoprint:
See [](https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint#bed-level-visualizer-support)Bed level visualizer support
But with own GCODE for Leveling in the Plugin.

**But i can confirm your issue.**

In my case the printer restarted after enabling auto level in EEPROM Manager in the past.
Now the printer doesn't hang up anymore when I activate it, but **it doesn't stay active**, you´re right.

mriscoc, 2022-04-13:
Any G28 / Homing will disable the auto leveling.
I just tested Octoprint:
```gcode
G28   ; Autohome
C29 N3 C0   ; Set 3x3 mesh and auto leveling bed temperature to 0 (only for test)
G29   ; Start auto leveling (for have a valid mesh)
M500   ; Save the mesh to the EEPROM
C108   ; Quit the Mesh Viewer
```
Now with a valid mesh in the EEPROM I go to the EEPROM editor Plugin
Press Load to load a valid set of parameters from the printer to the plugin
Go to the Autolevel section of the plugin
Check/Uncheck Enable and press Save
The z-offset icon changes color to confirm active/inactive auto leveling.

mriscoc, 2022-04-13:
Maybe not because in 10 minutes the printer disables the steppers motors then for do any movement safely it is necessary to do a home and any home disables the auto leveling.

daniel-volz, 2022-04-14:
> Maybe not because in 10 minutes the printer disables the steppers motors then for do any movement safely it is necessary to do a home and any home disables the auto leveling.

Thanks for your explanation, it's not a bug, it's standard in the Marlin software.
Maybe the whole thing will be added to the documentation, I didn't realize that this function would be deactivated by auto homing.

Does it maybe make sense to set **RESTORE_LEVELING_AFTER_G28** for the Ender 3 S1 in Marlin?
https://marlinfw.org/docs/gcode/M420.html

mriscoc, 2022-04-15:
RESTORE_LEVELING_AFTER_G28 could cause problems.

closed 2022-04-15 by mriscoc