mriscoc/Ender3V2S1 #117: Strange Scrolling on Ender 3 V2 Stock Display

andrewandneil, 2022-04-13:
### Description

Artefacts while scrolling through menu on Ender 3 V2’s stock LCD

### Steps to Reproduce

Opening any menu and scrolling causes scroll select bar highlight to ‘follow’ the current selection and instead of removing previous highlight, remains until scrolled on top of. It’s difficult to explain please see video below.

**Expected behavior:** Bars to no longer be highlighted when they are no longer selected

**Actual behavior:** Bars remain highlighted when scrolling through the screen until scrolled back over at which point it is cleared

#### Additional Information

Printer was flashed with the 422 BLTouch pre compiled firmware

[image]

[image]

mriscoc, 2022-04-13:
Please post a capture of your info page. From your screenshot, something is bad with your version because the icons are not with transparent background. Maybe the screen is loaded with an incompatible icon set.

andrewandneil, 2022-04-13:
[image]
Is this what you were asking for?

mriscoc, 2022-04-13:
Did you flash your screen before? If yes, you need to re-flash with a compatible icon set from here: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

andrewandneil, 2022-04-13:
You know, I think you’re right. Doesn’t fit either the DACAI nor the DWIN screen type. Just got it a few days ago. Everything is stock apart from fans and a CRTouch install.
[image]

mriscoc, 2022-04-13:
Please take a closeup picture from the main chip of the board of the screen unit, as clear as possible, and also from the labels to do a research

andrewandneil, 2022-04-13:
[image]
[image]
[image]

Hope these help. Took them zoomed it as much as I could. Please notify if you require more. Thank you for all this.

andrewandneil, 2022-04-14:
So I just got a chance to try it out. This has fixed the strange scrolling issue, however, scrolling seems ‘laggy’. But much more useable than before. However, I don’t know if you tried to, but it has not fixed the black boxes around the icons. Please see video attached. Thank you again
https://user-images.githubusercontent.com/20885631/163368531-d5d8d2a4-555c-42c3-b9ac-0db2c043a54b.MOV

mriscoc, 2022-04-15:
The icons are not fixable for your screen firmware version. You can select a darker background color in the Advanced Settings menu.

closed 2022-04-15 by mriscoc

mriscoc, 2022-10-18:
Hi, do you have the problem reported here?
https://github.com/mriscoc/Ender3V2S1/issues/323