mriscoc/Ender3V2S1 #119: Feature Request: Choose corner for Tramming Reference.

Moobien, 2022-04-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Often when I am tramming by bed I will be prompted to tram my rear left corner, I adjust other corners instead as my rear left has the wiring harness so I try to adjust it as little as possible.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
The ability to choose a corner to use as a reference for tramming against.  If corner X is chosen as a reference corner, the other 3 corners will tram against it instead of against each other.

SmallTownSound, 2022-04-21:
+1 for that, please add it.

mriscoc, 2022-05-29:
Currently the reference is the average of the four corners, that allows faster tramming, I recommend starting with the other corners above the problematic one.

closed 2022-05-29 by mriscoc

SmallTownSound, 2022-05-29:
Not a problematic corner.  The left back corner should always be used as a reference since it has the least travel, and should always be used as the reference. Sent from Mail for Windows From: Miguel Risco-CastilloSent: Sunday, May 29, 2022 6:06 PMTo: mriscoc/Ender3V2S1Cc: Don Turnock; CommentSubject: Re: [mriscoc/Ender3V2S1] Feature Request: Choose corner for Tramming Reference. (Issue #119) Currently the reference is the average of the four corners, that allows faster tramming, I recommend starting with the other corners above the problematic one.