mriscoc/Ender3V2S1 #120: JPEG Thumbnail not visible

Transcedental, 2022-04-16:
Printer Ender 3  V2, Firmware 20220330, firmware of display updated (DACAI), Cura v. 4.13.1 - plugin "Create JPEG Thumbnail" installed and activated. 
When Cura plugin is deactivated and new STL file sliced, then printer show QR code with link - OK from my side.
When Cura plugin is active then Thumbnail is generated in G-code but in printer is not visible.

[image]
[image]
[image]
[image]

Nbrooks639, 2022-04-16:
[image]

KriswiffaK, 2022-04-18:
Same deal here-

However it used to work

mriscoc, 2022-04-18:
> Same deal here-
> 
> However it used to work

If the embedded thumbnail is too large the preview is disabled.

KriswiffaK, 2022-04-18:
> > Same deal here-
> > 
> > However it used to work
> 
> If the embedded thumbnail is too large the preview is disabled.

230x180 set in cura

mriscoc, 2022-04-18:
> > > Same deal here-
> > > However it used to work
> > 
> > 
> > If the embedded thumbnail is too large the preview is disabled.
> 
> 230x180 set in cura

Not only about the dimensions, but thumbnail size in bytes.

closed 2022-04-19 by mriscoc