mriscoc/Ender3V2S1 #124: Problems with printing gcode files without preview

clanganke, 2022-04-18:
When I start printing a  gcode file without a preview on an Ender 3 S1, the confirmation screen for the print is being messed up.

[image]

In #121 it is said that a QR code is being displayed in this situation, and that this was ok, but I would consider this output not as being a correct QR code, and worse, in all following screens the background of the upper status line is not being reset: Here I returned to the main screen:

[image]

I use Ender3S1-20220330 on an Ender 3 S1 with STM32F103RCT6 chip. I migrated from Ender-3s1-hw24s1_301-sw-V1.0.5_C-CRTouch with applied resource bundle.

mriscoc, 2022-04-18:
Your screen font is incompatible with QR codes: https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview#qr-code-screen-glitches

clanganke, 2022-04-18:
Thnx for the quick response. Sorry - I already viewed this page

https://github.com/mriscoc/Ender3V2S1/wiki/How-to-update-the-display

but didn't scroll it down to the very bottom, would have reckognized the screenshot otherwise :-o I think I bailed out after having read the part about Cura that I needed...

However, this page says:

> If you have the stock icons and display assets you don't need to update the display to use this firmware. 

I already applied the **private**  folder of the  Ender-3s1-hw24s1_301-sw-V1.0.5_C-CRTouch firmware when I got the printer two month ago.  Just for safety I performed that again, but the error persists. And AFAIS it is binary identical to the "display assets/stock/private" folder of your repository.

So I think I use "stock icons" and "a compatible display asset", or am I wrong? Do I have to use sth else?

Thnx alot for yout support, and for sure for your work on this great product!

mriscoc, 2022-04-18:
Thanks, 
If it seems similar, please check if the font file is different.

clanganke, 2022-04-19:
Here the result of a binary comparison of the private folder of the stock firmware and your current release, where the color black means "no difference" (with differences files would display in red):

[image]

Or is there sth else that I missed to check?

mriscoc, 2022-04-19:
Thanks, can you verify that your screen is a DACAI unit? Currently there is three different screens.

mriscoc, 2022-04-19:
ok, thanks for your confirmation, in the test group of Telegram there are people with DACAI screens and they don't have that QR incompatibility, so, maybe your display has a firmware that is not compatible with that feature. As a last attempt try flashing Giadej's icon set. You can get rid of that dialog if you compile the firmware disabling the HAS_PREVIEW feature.

clanganke, 2022-04-19:
Unfortunately the other icon set did not help the problem. But it looks nice so I'll keep it with the stock firmware for now :-)

Just for my understanding:  the QR code gets displayed there if there is no preview in a g-code file, so what useful information can be provided by it in this case? 

I am curious about building the project, but I wonder if that is doable for me, not being an experienced developer under Linux. I know platformio a bit from building Tasmota - under Windows though - and I know only little about Docker.  May it be that is was only firing off a platformio command and everything else just happens (like with Tasmota ...)?

Thanks a lot for the support!

mriscoc, 2022-04-19:
I use Platformio in Visual Studio Code on Windows, compiling the firmware is the easy part of the process, customizing it is a little more difficult.
The QR shows a URL to the guide to enable the slicer to generate the preview.

mriscoc, 2022-04-19:
For remove the glitches, go to the Advanced Menu and turn off/on the LCD background. Also in the tune menu.

mriscoc, 2022-04-20:
It was not removed for the S1 because some people like to have the printer details and don't take care about the missing preview, something as this:
[image]

clanganke, 2022-04-20:
Oh sure, I very much agree with that, I find that text very valuable as well.  I only meant to not display the QR code, and only now I realized that I did not understand that disabling HAS_PREVIEW would skip the whole dialog. So a remaining question is if it is possible or not to only leave the QR code out for the Ender 3 S1 variant. If yes, maybe you consider doing that, if not, I would not vote for disabling the dialog. Will scan the release notes from time to time, I think ;-)

So it is just a problem for me (at least currently !?) , for whatever and unknown reason. As a workaround I still can include previews, even if I can't see them. Or possibly even patch only the smallest possible dummy preview into my gcode files - I already tried that manually and it worked. Both are definitely not the best solution, but doable.

I would consider that case closed then, again thanks a lot for your time and for your work on this project!

closed 2022-04-20 by clanganke

clanganke, 2022-04-24:
Just found that with Ender3S1-20220419 at least the corrupted background is being reset after the print confirmation dialog - thanx a lot! This is well good enough when I accidentally hit a gcode file not containing a preview.

Other than that my powershell script adds a dummy preview to the gcode file being generated from stl by CuraEngine, so gcode generated by OpenSCAD / CuraEngine works for now!

Just for curiosity - is there any chance getting Cura post processing to run within the CuraEngine process when calling it from commandline?
Possibly there is a JSON fragment that could be incorporated into the copy of the printer definition file, like it is necessary for centering the object, as described in 
   https://3dprinting.stackexchange.com/questions/4792/centering-model-with-curaengine-when-creating-g-code
? Other than that I could not find any overview on JSON fragments that would be used by CuraEngine, if there is any at all...

mriscoc, 2022-04-25:
I don't know too much about CuraEngine, but maybe you must define your script to be loaded into a plugin throught some parameters in the .ini configuration file that you pass in the command line.