mriscoc/Ender3V2S1 #125: Changed settings are not stored to EEPROM when saved (STM32F401)

theSt33v, 2022-04-19:
**Describe the bug**
Changed settings are not stored to EEPROM when saved.

**To Reproduce**
Steps to reproduce the behavior:
1. Change any setting
2. Select "Store Settings" or send gcode M500 through terminal.
3. Power cycle printer
4. All settings have reverted to firmware defaults.

**Expected behavior**
Changed settings should survive a power cycle after being stored.

**Version (please complete the following information):**
Experimental build 2.1.0.1 15Apr2022 (STM32F401).

mriscoc, 2022-04-19:
Thank you for your report

mriscoc, 2022-04-20:
Please try the latest version: https://github.com/mriscoc/Ender3V2S1/issues/128

bullitt186, 2022-04-20:
Unfortunately, the Issue is still unresolved with Experimental-Ender3S1-STM32F4-20220420.

mriscoc, 2022-04-20:
ok, thank you, I'll look deeper later.

rockstarbill, 2022-04-22:
Just confirming, the issue persists in the latest build as well.

airs0urce, 2022-04-22:
Yes, same for me, after I restart Ender 3 S1 all settings reverted to defaults

wamsuperman, 2022-04-24:
I am having an issue it is storing the settings but not actually using them since newest firmware I like the toolbar but it is costing me on not being able to up my acceleration

theSt33v, 2022-04-25:
I also cannot store settings with the latest build. I read on the Facebook group today that someone with an F4 board does not have this issue. It's so strange that the behavior is so inconsistent with boards that have identical hardware. Maybe the path to the EEPROM is different depending on the board, which is why the save is unsuccessful in some cases?

theSt33v, 2022-04-25:
> I am having an issue it is storing the settings but not actually using them since newest firmware I like the toolbar but it is costing me on not being able to up my acceleration

Are you sure that your slicer is not defining the acceleration parameters?

jeeltuss, 2022-04-25:
Hello, just want to post this hoping it can help to shed light on the issue.
I was fiddling with the experimental firmware & octoprint, while I was testing the missing store of settings I used the "Reboot printer" function in Control section.
After power cycling the printer reported "M112 Printer Killed", and asked to be manually shut down.

Then the web interface of octoprint reported this:

`Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: EEPROM CRC mismatch - (stored) 34127 != 59449 (calculated)!`

thesnarkyone, 2022-04-25:
I am experiencing the same issue as well. Power cycles remove settings.

jeeltuss, 2022-04-25:
> > Hello, just want to post this hoping it can help to shed light on the issue. I was fiddling with the experimental firmware & octoprint, while I was testing the missing store of settings I used the "Reboot printer" function in Control section. After power cycling the printer reported "M112 Printer Killed", and asked to be manually shut down.
> > Then the web interface of octoprint reported this:
> > `Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: EEPROM CRC mismatch - (stored) 34127 != 59449 (calculated)!`
> 
> Check error handling in https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

Oh ok, I was hoping that the CRC mismatch error can give a clue on why the settings are not saved. My bad. Thanks for the link.

theSt33v, 2022-04-28:
The issue still exists in the 27Apr build.

vb138, 2022-04-28:
Confirmed, I have the same problem.

bankm1, 2022-05-03:
+1 settings will not save

airs0urce, 2022-05-03:
As I have access to physical machine just want to let you know if you need some details to solve this issue (photos of chips etc) I can send them

mriscoc, 2022-05-03:
> As I have access to physical machine just want to let you know if you need some details to solve this issue (photos of chips etc) I can make and send them

Thank you, I do dozens of fast tests before the release of a version.

fabiomaticus, 2022-05-05:
The issue still exists in the 4 May build.

mriscoc, 2022-05-05:
> The issue still exists in the 4 May build.

Thanks for the feedback

vb138, 2022-05-06:
For interest, please visit https://www.th3dstudio.com/hc/downloads/unified-2-firmware/creality/creality-boards-creality/creality-ender-3-s1-pro-firmware-v24s1_301-board/ and https://github.com/th3dstudio/UnifiedFirmware/commits/2.0.x

jeeltuss, 2022-05-07:
> For interest, please visit https://www.th3dstudio.com/hc/downloads/unified-2-firmware/creality/creality-boards-creality/creality-ender-3-s1-pro-firmware-v24s1_301-board/ and https://github.com/th3dstudio/UnifiedFirmware/commits/2.0.x

I am currently running the TH3D Unified fimware on my S1 (non Pro) with the STM32F401 chip, and I can confirm that all settings I tried are correctly saved when power-cycling the printer. Maybe this can help @mriscoc in investigating the issue.

theSt33v, 2022-05-07:
I did a little bit of investigation and I think I need to clarify the issue. The settings are, in fact, stored correctly. If you save settings to EEPROM and then restore default settings, you can easily restore your previously-saved settings. This may indicate that the issue is not that the settings are not saved, but rather they are being wiped at boot.

mriscoc, 2022-05-07:
> I did a little bit of investigation and I think I need to clarify the issue. The settings are, in fact, stored correctly. If you save settings to EEPROM and then restore default settings, you can [...]

Yes, the issue is with the calculus of the CRC EEPROM value in the F4 chips.

mriscoc, 2022-05-07:
If someone is confident with coding and compiling the firmware and has the F4 board to do test then they can try to check these relevant parts of the source:

https://github.com/mriscoc/Ender3V2S1/blob/c2744e01e00ca3d97800a91d7c3aa969bb9ae1c2/Marlin/src/HAL/STM32/eeprom_bl24cxx.cpp#L72-L82

https://github.com/mriscoc/Ender3V2S1/blob/c2744e01e00ca3d97800a91d7c3aa969bb9ae1c2/Marlin/src/module/settings.cpp#L1693-L1694

and

https://github.com/mriscoc/Ender3V2S1/blob/c2744e01e00ca3d97800a91d7c3aa969bb9ae1c2/Marlin/src/module/settings.cpp#L2610-L2616

theSt33v, 2022-05-07:
There's not enough flash space to enable that for the F4 without cutting other things. I'm trying to disable some features to make room but I'm running into compile errors because the features are "required for proui." I'll keep at it, but if you know any features I can cut without breaking things too much, that would be helpful.

mriscoc, 2022-05-07:
Comment out in dwin_defines.h the line `#define ProUIex 1`
That will give you flash space.

theSt33v, 2022-05-07:
Doing that caused build errors, but I was able to get a successful and functional build by disabling M503 in configuration.h, setting HAS_GCODE_PREVIEW to 0 in proui.h and commenting out the error if for HAS_GCODE_PREVIEW in dwin_defines.h. I'll see what I can find out with this crc problem.

jeeltuss, 2022-05-07:
> Doing that caused build errors, but I was able to get a successful and functional build by disabling M503 in configuration.h, setting HAS_GCODE_PREVIEW to 0 in proui.h and commenting out the error if for HAS_GCODE_PREVIEW in dwin_defines.h. I'll see what I can find out with this CRC problem.
> 
> 
> 
> Could you give me some more background on what is known so far about the problem? How do we know that the issue is CRC calculation related? So far, I haven't been able to trigger any of the CRC errors.

As I was saying [here](https://github.com/mriscoc/Ender3V2S1/issues/125#issuecomment-1108864947), I had octoprint reporting errors from the printer regarding crc mismatch. 
I remember saving settings straight from the printer after modifying any of them (i.e. the z offset), while octoprint was connected. Hope this helps.

mriscoc, 2022-05-08:
> As I was saying [here](https://github.com/mriscoc/Ender3V2S1/issues/125#issuecomment-1108864947), I had octoprint reporting errors from the printer regarding crc mismatch. I remember saving settings [...]
Yes, you were in the correct way.

mriscoc, 2022-05-08:
Of course, it is possible to disable the crc checking but it could be dangerous, but maybe it is a simple test to check if only the crc is the problem.

theSt33v, 2022-05-08:
I just did that and it is indeed the problem. If you change eeprom_error=true to false, settings are preserved after power cycle.

mriscoc, 2022-05-08:
So, now we need to understand why only for the F4 is `working_crc != stored_crc`

theSt33v, 2022-05-08:
Right. Your code in that section is the same as Creality source, and stock firmware does not have this issue, so it's very strange.

mriscoc, 2022-05-08:
Creality source is based on STM32 Maple I think, that uses a different HAL implementation. The Marlin bugfix branch has deprecated the Maple version and uses STM libraries.

theSt33v, 2022-05-09:
I did a lot of poking around and I only ever get that CRC error at boot. All other commands that trigger CRC checks (M500 and M501) are always successful and return CRC values that are consistent and change when they're supposed to. Even when I disable the automatic factory reset (M502) that the error usually triggers, the CRC checks pass. This means that even when you do literally nothing to the flash, the CRC checks pass after boot.

It seems to me that only the CRC check that happens at boot is problematic, and it's a false positive. Is there a different section of code that governs the CRC check at boot? Are the working_crc and/or stored_crc values incorrectly assigned at boot due to an incorrect hardcoded value somewhere in the settings, but correctly assigned when triggered post-boot? I don't know, but I'm going to just disable the error in the builds that I use. I'd like to get to the root of this, but I think I've reached the limit of my very limited ability to diagnose this stuff. All I can say is all the relevant sections of code are nearly identical to their equivalent sections in the TH3D unified firmware (which doesn't suffer from this issue), so I don't think the problem is there.

mriscoc, 2022-05-10:
But why does that error only occur for the F4 SoC? Unfortunately, I don't have an F4 board to do debugging.
From your tests I can summarize:
M500 Save data to EEPROM and store a correct calculated (working) CRC
M501 Read data from EEPROM and the stored CRC is equal to calculated (working) CRC
M997 (Reboot) triggers the error `working_crc != stored_crc`, but what CRC is bad? the working or the stored, or both?

theSt33v, 2022-05-10:
That's correct. As for which one is bad, it wasn't consistent in my testing. One of them always matched the CRC from before the reboot (which makes it the correct one), but sometimes it was working and sometimes it was stored. I'll do more systematic tests today to see if there is a pattern.

mriscoc, 2022-05-16:
Fixed in the last release candidate version https://github.com/mriscoc/Ender3V2S1/issues/128#issue-1209999797

vb138, 2022-05-16:
Tested. Now all settings are preserved even after rebooting the device. Thank you.

mriscoc, 2022-05-16:
> Tested. Now all settings are preserved even after rebooting the device. Thank you.

Thank you for your confirmation. I'm closing the issue now.

closed 2022-05-16 by mriscoc