mriscoc/Ender3V2S1 #128: Experimental Support for Ender 3 S1 with the STM32F401RCT6 SoC

mriscoc, 2022-04-20:
New Experimental version with the changes from the APRIL2022 release:  

[Experimental-Ender3S1-STM32F4-20220420.zip](https://github.com/mriscoc/Ender3V2S1/files/8523879/Experimental-Ender3S1-STM32F4-20220420.zip)

@houseofbugs has discover that the firmware must be inside the STM32F4_UPDATE folder of the memory card for a correct flash procedure in some S1 F4 printers.

This is a private build for testing the preliminary compatibility of the firmware with the **Ender3S1** with the **F4 SoC**, please use it with caution.

### Disclaimer
THIS FIRMWARE AND ALL OTHER FILES IN THE DOWNLOAD ARE PROVIDED FREE OF CHARGE WITH NO WARRANTY OR GUARANTEE. SUPPORT IS NOT INCLUDED JUST BECAUSE YOU DOWNLOADED THE FIRMWARE. WE ARE NOT LIABLE FOR ANY DAMAGE TO YOUR PRINTER, PERSON, OR ANY OTHER PROPERTY DUE TO THE USE OF THIS FIRMWARE. IF YOU DO NOT AGREE TO THESE TERMS THEN DO NOT USE THIS FIRMWARE.

mriscoc, 2022-04-27:
Zip file updated

mriscoc, 2022-05-05:
Zip file updated, applied latest Marlin Bugfix.

kiby, 2022-05-05:
Hi.
Experimental-Ender3S1-STM32F4-20220504 Tested.
Changed settings are not stored to EEPROM when saved (STM32F401)

WZ9V, 2022-05-09:
I can also confirm that settings are not saved.  I just got a replacement board which has the F4, ran the Hotend PID tune, saved the updated settings, powered down and upon restart the setting were back to the defaults.

Maxtor78, 2022-05-09:
hello, 
so me when I install it the screen is frozen on the logo.

WZ9V, 2022-05-09:
Did you upgrade the screen firmware first?  If not, you will need to open it up and upgrade it with a micro SD card.  Look at Wiki topic updating the display.

nickgarvey, 2022-05-09:
I'm not sure this works for the pro version?

WZ9V, 2022-05-10:
Oh yeah, I forgot that I swapped the non-touch screen display with the knob on my S1 Pro.  The touchscreen won't work.

shmink, 2022-05-14:
Thanks for this. My printer wasn't even booting until I tried this firmware so you really saved me.

As others have mentioned it does seem like some settings aren't being saved. Should there be a _save config_ in the prepare section?

mriscoc, 2022-05-14:
> Thanks for this. My printer wasn't even booting until I tried this firmware so you really saved me.
> 
> As others have mentioned it does seem like some settings aren't being saved. Should there be a _save config_ in the prepare section?

The is currently a known issue https://github.com/mriscoc/Ender3V2S1/issues/125 we are working to fix that.

mriscoc, 2022-05-16:
New Release Candidate for the Ender3S1F4, the file has been updated.

shmink, 2022-05-16:
Super small detail that I've noticed since updating my firmware to this new version is that 9x9 mesh levelling seems to fail all the time for me. Not sure if it's just me though.

WZ9V, 2022-05-16:
Mine finished the 9x9 grid but the front left was squished and the back was to high.  I trammed the bed so that it was within 0.01 on the corners.

WZ9V, 2022-05-16:
> New Release Candidate for the Ender3S1F4, the file has been updated.

Link?

TopazAaron, 2022-05-17:
@mriscoc Appreciate the work on the RC for the F4 chip. I've been playing around with it and I've noticed the filament runout sensor isn't working still. just letting you know.

mriscoc, 2022-05-17:
> @mriscoc Appreciate the work on the RC for the F4 chip. I've been playing around with it and I've noticed the filament runout sensor isn't working still. just letting you know.

Check your runout distance and the End-stop diagnostic.

xXwinsXx, 2022-05-19:
Hi, ok the flow is not saved after the printer is switched off, which I don't like, the Z offset I set is retained after the switch is switched off, but the first layer is still not always the same, what I noticed is that the Z motors in the print don't move which means to me that BED leveling isn't carried over, that's a real shame.

mriscoc, 2022-05-19:
> Hi, ok the flow is not saved after the printer is switched off, which I don't like, the Z offset I set is retained after the switch is switched off, but the first layer is still not always the same, [...]

The flow is never saved, not even in the stock firmware, flow is only a corrective parameter when the calibration of the e-steps is not ok. Don't forget to add the M420 S1 after the G28 as is indicated in the wiki.

mriscoc, 2022-05-19:
I'm closing this issue now, please open a new issue if you found a bug or want to request a feature, remember to do a search first on the closed issues for not to duplicate them.

closed 2022-05-19 by mriscoc

trulow, 2022-05-21:
Will the latest release [Ender3S1-STM32F4-20220515.zip](https://github.com/mriscoc/Ender3V2S1/files/8697054/Ender3S1-STM32F4-20220515.zip) work with S1 Pro and the 300 degree all metal hot end?

trulow, 2022-05-23:
> > Will the latest release [Ender3S1-STM32F4-20220515.zip](https://github.com/mriscoc/Ender3V2S1/files/8697054/Ender3S1-STM32F4-20220515.zip) work with S1 Pro and the 300 degree all metal hot end?
> 
> nope, only 260°... maybe we get a new 300° version from @mriscoc :) :)

Thanks, I was able to get this version of the firmware working on my S1 Pro. Had to purchase and install a v2 screen but everything seems to work without issues. Hopefully there's an update soon to which supports the higher temps the S1 Pro provides.

acothebraco, 2022-05-25:
> Thanks, I was able to get this version of the firmware working on my S1 Pro. Had to purchase and install a v2 screen but everything seems to work without issues. Hopefully there's an update soon to [...]

Edit the Configuration.h on HEATER_0_MAXTEMP 275 to 295 and recompile the Firmware again and you have higher Temp on der Extruder Pro.

GS1305, 2022-05-26:
Hallo alle zusammen. Ich haffe das einer von euch mir weiter helfen kann. Meine Situation ist das ich nach dem Tausch meines Defektem Mainboard das neue nicht über SD karte(4GB-32GB FAT32) nicht flashen kann. Egal welche Release ich benutze. Nun habe ich mir Arduin uno besorg und möchte nun direkt die Firmware und den Bootloader Flashen. Leider finde ich keine Information welche pins ich verbinden muss. 4 pins STM32 GND SWDIO SWCLK und VCC. Frage wo muss ich SWDIO und SWCLK am Arduino anschließen. Oder het das auch über AREF (Arduino) USB STM32?

Skydsler, 2022-05-26:
> Hallo alle zusammen. Ich haffe das einer von euch mir weiter helfen kann. Meine Situation ist das ich nach dem Tausch meines Defektem Mainboard das neue nicht über SD karte(4GB-32GB FAT32) nicht [...]

Hallöchen,

Hast du schon geschaut welcher Chip auf dem neuen Board ist?

GS1305, 2022-05-26:
STM32F103RCT6

Skydsler, 2022-05-26:
> STM32F103RCT6

Dann sollte die hier ( https://github.com/mriscoc/Ender3V2S1/releases/download/20220419/Ender3S1-20220419.bin)  im root Verzeichnis der SD funktionieren... Ggf auch die MM variante

GS1305, 2022-05-26:
alles schon probiert wie es scheint worde beim aufspielen der Firmware das annehmen von neuer über SD ausgeschaltet. Dadurch liest das System nicht ob eine neue Version vorhanden ist und installiert sie. Um das zu beheben habe ich herausgefunden muss ich einen neuen bootloader aufspielen und dann sollte es auch mit SD funktionieren. Leider hat das Neue Systemboard nur 4 pin und USB anschluss. Daher meine Frage wie schliesse ich das STM an Arduino an. Für 8 Bit finde ich alle informationen aber nichts zum 32Bit

Skydsler, 2022-05-26:
Klingt schon komisch. Wo hast du dein board bestellt? Habe jetzt 4 Stück bestellt und immer ein F401 board bekommen... Hätte auch gern ein 103er :)

GS1305, 2022-05-26:
Das alte und das neue sind identisch. Hab gelesen das man den zugriff für Firmenware von der SD karte entsprechend ausschalten kann. Somit muss ich nur einen neuen Bootloader aufspielen und dann anschließend über USB die Firmware laden

xXwinsXx, 2022-05-26:
Ich habo vom Creality Support einen RAR Datei bekommen. Der Inhalt ist, Booloader samt Firmware für denn F4 Chip inklusive Anleitung.

GS1305, 2022-05-26:
OK kannst du mir die per mail senden

xXwinsXx, 2022-05-26:
Wenn ich die hier hoch lade haben alle was davon. Ich weiß nicht wie das mit nem Iphone funktioniert. Du?

Skydsler, 2022-05-26:
Da müsstest du mit @mriscoc reden, ob er die Datei mit veröffentlichen will.

GS1305, 2022-05-26:
hei xXwinXx kannst du sie mir mach gsvws1305 @ gmail . com senden. Vieleicht ist da auch ne anleitung für die anschlüsse drin.

xXwinsXx, 2022-05-26:
Ok gut, dann hoffe ich er wird es lesen.  Gib mir eine Email

GS1305, 2022-05-26:
ok danke

mtigeli, 2022-05-26:
Hallo
Haben Sie versucht, die Firmware mit hilfe dieser Video-Tutorials zu flashen?
[https://youtu.be/byOzpJXpUTo?t=39](url)

GS1305, 2022-05-26:
3.0.3

GS1305, 2022-05-26:
mtigeli das ist nur die firmware für display

xXwinsXx, 2022-05-26:
> mtigeli das ist nur die firmware für display

Nein das ist die Firmware fürs Board. Die fürs Display ist aber auch dabei

GS1305, 2022-05-26:
ok dann werde ich mal es versuchen

GS1305, 2022-05-26:
Sorry [xXwinsXx] es war für mtideli. Von dir habe ich noch nichts erhalten hatte dir noch mal die Adresse gesendet gsvws1305@gmail.com

xXwinsXx, 2022-05-26:
> Sorry [xXwinsXx] es war für mtideli. Von dir habe ich noch nichts erhalten hatte dir noch mal die Adresse gesendet [gsvws1305@gmail.com](mailto:gsvws1305@gmail.com)

Immer noch nicht? Spam Ordner?

GS1305, 2022-05-26:
angekommen. War im Spam wie lange hat das bei dir gedauert

Skydsler, 2022-05-26:
Dann war es aber doch kein f103 Board oder?

GS1305, 2022-05-26:
doch

GS1305, 2022-05-27:
Nun mal wieder eine frage an alle. Leider scheint es so das es kaum mehr möglich ist eine Marlin Firmware auf das STM32F103RCT6 zu bringen. Habe nun eine neues von Creality aufspielen können aber kein von Marlin. Habe alles mögliche versucht. Der Bootloader hängt sie immer wieder auf und ich muss dann mit dem Bootloader und dem Update von  Creality drüber gehen. Kann es sein das ich nur über usb c - icsp (zb Arduino) flashen kann. Oder hat da jemand ne tipp. Würde gern wieder mit Marlin abeiten.

GS1305, 2022-05-28:
So neues update von mir nach langen hin und herhabe ich es geschaft mein neues board upzudaten. Zu meiner Schande muss ich gestehen das ich davon ausging ein 103 bekommen zu haben. So wie es aber jetzt aussieht habe ich ein F4 bekommen. Nach dem ich das Verzeichnis auf SD erstellt habe und die Datei dort hinein kopiert hatte ging das update ohne Probleme durch. Jetzt habe ich nur noch die Herausforderung wie ich meinen neuen Extruder(300 grad) richtig aufheizen kann. Da warte ich sehnsüchtig auf ein neues Update.

mriscoc, 2022-05-28:
For unlock the temperature limit of the hotend you can use this G-code in a terminal as Octoprint with the latest firmware version:
C104 U1 T300  ; set absolute limit to 300°C
M500               ; save settings
But be careful to not to exceed the manufacturer's limit.

Skydsler, 2022-05-28:
> So neues update von mir nach langen hin und herhabe ich es geschaft mein neues board upzudaten. Zu meiner Schande muss ich gestehen das ich davon ausging ein 103 bekommen zu haben. So wie es aber [...]

Wenn du magst kann ich dir eine Firmware compilen für 300°

GS1305, 2022-05-28:
Das wäre super.
Werde mich in Zukunft ein wenig mehr mit der Programmierung auseinander setzen finde die Möglichkeiten die wir mit ne 3d Drucker haben echt super.  Was mir immer fehlt im Display ist die angaben zu den Einheiten. z.b ob mm, Stepps usw.
Gerade beim einstellen vom Druckbett arbeite ich mit Messschieber und ner Wasserwaage um das Bett besser auszurichten. Hab feststellen müssen das die Grundplatte nicht wirklich plan ist. Werde mir wahrscheinlich mal eine machen lassen die wirklich 100% eben ist. Ich bekomme es nie hin das alles grün ist. Die ecken Toleranz ca 0,2 zur Mitte +- 8. Habe in der Mitte mit Papier etwas aus geglichen danach -+ 3

trulow, 2022-05-29:
@mriscoc Thanks, I just updated my S1 Pro with the latest version, sent the custom G-code you provided and everything looks good. I appreciate your help with all of this!

neumo, 2022-05-29:
> Wenn du magst kann ich dir eine Firmware compilen für 300°

Da hab ich auch Interesse. Ich bin leider zu blöd die neue Firmware für den EnderS1F4 zu finden :-(

GS1305, 2022-05-30:
Jep das wäre auch ne gute Sache mit 110 Grad. Hätte da noch ne Kleinlichkeit. Wäre es möglich 2 oder 3 Profile zu lager um sie dann abzurufen. Denke gerade ans leveln und div unterschiedliche druck platten. Habe eine von Crality und eine von Cura. Bedie sind unterschiedlich dick. Wäre schön wenn man beim wechseln einfach das entsprechende Profil auf rufen kann und dann auto levelt.

neumo, 2022-05-31:
mmh  so uneben ist das bei Dir? 0.8 mm finde ich schon enorm viel!? Ich hab neulich die Glasplatte vom alten A6 auf den S1 geklemmt - a Traum! Aber selbst Die hat gewisse Unebenheiten, doch die sind im Bereich der 2. Nachkommastelle und eher klein. 

Gibt es eigentlich irgendwo eine Anleitung, wie man die Firmware selbst kompiliert? Hatte gesucht, aber nicht wirklich was dazu gefunden...

neumo, 2022-05-31:
> Achtung!!! die Version Ender3S!-F4-20220527 hat einen Fehler. Der Flow wird bei mir nach dem ersten Layer unterbrochen und wird auch nicht fortgesetzt Habe die Version Ender3S1-STM32F4-20220515 wieder drauf gemacht und alles wieder ok. Also schön aufpassen

Danke für den Test, ich wollte das heute Abend auch probieren. Hast Du Screens gemacht? Wenn Du das Issue reportest könnte das der Chef beheben. Falls nicht, kann ich das noch nachholen...sag Bescheid.

GS1305, 2022-05-31:
nein leider nicht aber hier das Resultat von einen Rechteck mit 0,4 mm 
[image]
t

GS1305, 2022-05-31:
und das selbe nach dem update zurück. Sorry hatte es schon in den Müll entsorgt das wegen die krummen wände. :)
[image]

neumo, 2022-06-01:
> Achtung!!! die Version Ender3S!-F4-20220527 hat einen Fehler. Der Flow wird bei mir nach dem ersten Layer unterbrochen und wird auch nicht fortgesetzt Habe die Version Ender3S1-STM32F4-20220515 wieder drauf gemacht und alles wieder ok. Also schön aufpassen

Habe eben auf die Version F4 20220527 upgedated, Restore settings gemacht, Grund Setup gemacht, aber ohne PID Tuning. !. Druck läuft, bis jetzt Layer 13 , 2.6mm, keine Probleme mit dem Flow. Ich hab einen S1 Pro....keine Ahnung, warum das bei Dir nicht tut.
Gefühlt nimmt er aber das Mesh nicht her. :-(

GS1305, 2022-06-04:
Hei mriskoc wie lange brauchst bis ne Version für 300 Grad fertig ist. Wäre schön wenn ich meinen neuen Kopf mal testen könnte

[5 more replies]