mriscoc/Ender3V2S1 #137: Move or Copy Leveling Mesh size command

lrahnz, 2022-04-29:
I have a BLTouch.  I do my initial bed leveling using 2x2 mesh points to get the corners adjusted.  Then I do 9x9 mesh points for production use.  It would be easier if I could set the mesh points in the Prepare group of commands instead of going back and forth between the Prepare and Advanced menus

mriscoc, 2022-04-29:
In the latest versions there is not a 2x2 mesh any more. In the "Prepare" menu there is a bed tramming that takes measurements of the level of the corners to take average and suggest an adjustment of the leveling knobs.

lrahnz, 2022-04-29:
I loaded your latest firmware and the Bed Tramming wizard automates exactly what I was doing.  Very nice!  
 
I also like the configurable Toolbar.  This will be very helpful with day to day use.  Right now I have a “Menu Card” that I use to find my most frequently used commands.  Now I can put them on the toolbar and have them on screen.
 
Anyway, consider my issue closed.
 
Regards
Logan

closed 2022-04-30 by mriscoc