mriscoc/Ender3V2S1 #138: G29 Command not being executed

Moobien, 2022-04-30:
**Describe the bug**

since updating to the version released 10 days ago, I have noticed that my printer no longer performs G29 mesh levelling from gcode.  To test I also added M420 to my start gcode and not only does the printer not run the mesh level as it should with G29 it is also not loading the stored mesh from the M420.  Have a stored mesh that I manually did via the firmware menus.  Have enabled 'Activate Levelling' numerous times and it never stays enabled and during prints the mesh indicator at zoffset in the main print screen never changes from white to red.

Prior to this I hadn't used the UBL version, only the BLT versions, could this be the issue?

**Version (please complete the following information):**
Ender3V2-422BLT-UBL-20220419

**Additional context**
Add any other context about the problem here.

Moobien, 2022-04-30:
EDIT: reflashed to Ender3V2-422BLT-20220419 and its working.

mriscoc, 2022-04-30:
> EDIT: reflashed to Ender3V2-422BLT-20220419 and its working.

For UBL version you must change the G29 for G29P1

closed 2022-04-30 by mriscoc

Moobien, 2022-04-30:
What is the difference between the 2 commands?