mriscoc/Ender3V2S1 #140: No serial with octoprint after update

Dilbao, 2022-05-03:
**Describe the bug**
After updating to latest firmware, I lost serial connection with printer.

Octoprint installed on Raspberry Pi 4.
Ender 3 v2 (with 422 board) connected to RPi with usb cable.

So far I tried;
- Using different usb cables (both as +5V connected and disconnected)
- Setting 115K baud on firmware (both on and off)
- Using different baud rates manually on octoprint
- Using older versions of the firmware

So far, I am not getting any connection with any combination above. I need help to diagnose this any further.

mriscoc, 2022-05-03:
Currently there are no communication problems between Octoprint and the latest firmware version.

Dilbao, 2022-05-03:
Well, sure. But I do have a problem with serial connection that is not suppose to happen currently. How could I find what is wrong with my setup?

mriscoc, 2022-05-04:
Maybe a bad USB board connector.

mriscoc, 2022-05-04:
This is not firmware bug.

Dilbao, 2022-05-04:
I flashed other firmwares to test it but connection problem only happens with this one. I don't know what is different with this one.

mriscoc, 2022-05-04:
Some hardware could be incompatible with this firmware.

Dilbao, 2022-05-05:
I compiled latest nightly Marlin from source but problem is still occuring. I can confirm that this is a Marlin issue.

Japander, 2022-07-31:
I can nail it down.

Its the meatpack plugin from Octoprint, causing the serial connection not being able to be established anymore. Which kinda makes sence, since M115 tells me meatpack is not enabled.

Probably someone switching firmware from Jyvers to here isnt aware of. Maybe mentioning it in docs?

mriscoc, 2022-08-01:
> I can nail it down.
>
> Its the meatpack plugin from Octoprint, causing the serial connection not being able to be established anymore. Which kinda makes sence, since M115 tells me meatpack is not [...]

Thank you, I added a mention about that incompatibility.

Dilbao, 2022-08-01:
Thank you, everything works fine after disabling Meatpack plugin from Octoprint.