mriscoc/Ender3V2S1 #141: Compiling error to use FIX_MOUNTED_PROBE

pedrofalmeida, 2022-05-03:
**Describe the bug**
I got an Ender 3v2 with a fixed probe, so I need to compile this firmware with some adjusts in the configuration file. 

**To Reproduce**
1. In Configuration.h comment #define PROBE_MANUALLY
2. In Configuration.h uncomment #define FIX_MOUNTED_PROBE
3. In Configuration.h uncomment #define Z_SAFE_HOMING
4. While compiling in VSCode I got the following error:
`Compiling .pio\build\STM32F103RC_creality\src\src\lcd\e3v2\proui\dwinui.cpp.o
*** [.pio\build\STM32F103RC_creality\src\src\inc\Warnings.cpp.o] Error 1
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void SetZOffset()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:2009:7: error: 'babystep' was not declared in this scope
 2009 |       babystep.accum = round(planner.settings.axis_steps_per_mm[Z_AXIS] * BABY_Z_VAR);
      |       ^~~~~~~~
*** [.pio\build\STM32F103RC_creality\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1`

**Screenshots**
[image]

**Version (please complete the following information):**
It works fine in Ender3V2S1-20220314, since that in every version I got this error.
For reproduce I was using the branch version Ender3V2S1-Released

pedrofalmeida, 2022-05-03:
Thank you @mriscoc! It works like a charm! :)

closed 2022-05-03 by pedrofalmeida

mriscoc, 2022-05-03:
You are welcome