mriscoc/Ender3V2S1 #144: Ender 3 S1 Pro/ Sprite Pro Upgrade

woodypride94, 2022-05-06:
I don't know if this is the right way to go about this, but I've been trying all day and I just can't get the firmware to compile correctly myself no matter how many guides I watch. I have upgraded my original S1 with the all metal Sprite Pro extruder and hotend and would like to increase the max nozzle temperature to 300C. Could a pro/high temp version be easily added? I would really like to learn how to just fix the issue myself, but I seem to have hit a wall in Visual Studio Code.

Nbrooks639, 2022-05-07:
https://github.com/mriscoc/Special_Configurations/releases

woodypride94, 2022-05-07:
> https://github.com/mriscoc/Special_Configurations/releases

Thank you so much! I will look into this tomorrow and see if I can get it working properly.

woodypride94, 2022-05-07:
Ok so this special configuration allows me up to 285C but still not 300C.

theSt33v, 2022-05-07:
There is an error in the configuration for this build. Max temp is set to 300 but in Marlin, this is the emergency shutoff temp, not the max reachable temp. For a max reachable temp of 300, this value should be changed to 315 in the config file.

woodypride94, 2022-05-07:
Gotcha, that makes  sense. I spent quite a bit of time trying to compile the firmware myself and the only thing I changed was maxtemp to 300C before building the firmware with the Auto Build Marlin extension on Visual Studio Code, but when I attempt to flash the .bin to my printer it just gets stuck on the splash screen (I did change the file name to something new). I'll start over and see if I can get it to work this time. Thank you for the information!

mriscoc, 2022-05-07:
Be careful, the absolute maximum is 300 °C and there is a 15°C security overshoot.
When you set a temperature there is an "inertia" the heat doesn't stop immediately, that's why there is an overshoot.
https://github.com/mriscoc/Ender3V2S1/blob/c2744e01e00ca3d97800a91d7c3aa969bb9ae1c2/Marlin/Configuration.h#L601-L609

theSt33v, 2022-05-07:
If you're getting stuck at the splash screen, you have bigger problems than temperature settings. Does your printer have an F1 or F4 chip?

woodypride94, 2022-05-07:
> Be careful, the absolute maximum is 300 °C and there is a 15°C security overshoot. When you set a temperature there is an "inertia" the heat doesn't stop immediately, that's why there is an [...]

So what do you recommend to achieve 300C at the hotend? maxtemp of 310-305 with an overshoot of 10-5? Or do you think I can set the maxtemp to 315 with an overshoot of 15 and just be careful and monitor it closely any time I print anything up to 300C?

> If you're getting stuck at the splash screen, you have bigger problems than temperature settings. Does your printer have an F1 or F4 chip?

I assume thins is something I need to actually open up the mainboard to find out which I have not done yet. I have however flashed a few different firmwares and have not had an issue yet. I will open it up to find out for sure.

Again, the only thing I had changed before was maxtemp without changing mintemp or temp_sensor_0 so that may have been the issue.

I'm now attempting to build the firmware with these  new values, but the .bin file is not showing up (or is being sent to somewhere it's not suppose to) and I haven't been able to find it.

theSt33v, 2022-05-07:
If your printer came with stock firmware 1.X.X, you have the F1. If 3.X.X, F4. Since you were able to flash custom firmware successfully without checking this first, you probably have the F1. 

As for what the overshoot should be, unfortunately Creality doesn't have a Pro-specific variant of the S1 source code released, so it's hard to know for sure, but other S1 firmwares that support the Pro seem to keep the 15 C overshoot regardless of the maxtemp value. Whatever you decide, though, just make sure your maximum desired selectable temperature = selectable temperature + overshoot. Also, make sure to change your thermistor preset value from 1 to 13.

woodypride94, 2022-05-07:
Ok the main issue I'm having now is after building the firmware with Auto Build is that I don't get a .bin file anywhere.

theSt33v, 2022-05-07:
This is an issue with your vscode installation, not this firmware. I suggest seeking advice on more generic Marlin-building forums, as this issue section is for discussion firmware-specific issues.

woodypride94, 2022-05-07:
Ok thank you. I really appreciate all the help you have given!

closed 2022-05-08 by mriscoc

mm108, 2022-11-16:
Wondering if some folks are up to sponsoring a upgrade to "fully" support the Ender-3 S1 Pro? When I say fully I mean being able to utilize the full temperature range (300°C) of the Sprite Pro extruder and the max bed temperature(110°C) like the stock firmware allows. I use my 3D printer for hobby purposes but I do have some PC filament and other engineering grade filament some of which requires 290°C to 300°C. I am using this [firmware](https://github.com/mriscoc/Special_Configurations/releases/tag/T13) at the moment and it works out well - it does hit 285°C after updating to the [special releases](https://github.com/mriscoc/Special_Configurations/releases/tag/T13). So if there are other folks out there wishing for a full S1 Pro support, should we band together to donate? Thanks.

theSt33v, 2022-11-16:
> 

Allow me to save you some money:

1. Learn to compile Marlin firmware from source. There are lots of videos on YouTube that cover this.
2. Download the source code for the professional firmware.
3. Download the "special configuration" for the F1 or F4 chip depending on what is appropriate for your printer.
4. Edit configuration.h and change the value for #define TEMP_SENSOR_0 from 1 to 61.
5. In the same file, change the values for #define HEATER_0_MAXTEMP (and all the other heater maxtemp values that come after it) from 300 (or whatever it is) to 315.
6. Compile the firmware and flash the bin.

mriscoc, 2022-11-17:
> Wondering if some folks are up to sponsoring a upgrade to "fully" support the Ender-3 S1 Pro? When I say fully I mean being able to utilize the full temperature range (300°C) of the Sprite Pro extruder and the max bed temperature(110°C) like the stock firmware allows. I use my 3D printer for hobby purposes but I do have some PC filament and other engineering grade filament some of which requires 290°C to 300°C. I am using this [firmware](https://github.com/mriscoc/Special_Configurations/releases/tag/T13) at the moment and it works out well - it does hit 285°C after updating to the [special releases](https://github.com/mriscoc/Special_Configurations/releases/tag/T13). Their sprite pro extruder seems to be the stock on a few of their new machines.
  
Use the special custom g-code C104 to unlock maximum hotend temperature, for example:
`C104 U1 T315`

mm108, 2022-11-17:
> > 
> 
> Allow me to save you some money:
> 
> 1. Learn to compile Marlin firmware from source. There are lots of videos on YouTube that cover this.
> 2. Download the source code for the professional firmware.
> 3. Download the "special configuration" for the F1 or F4 chip depending on what is appropriate for your printer.
> 4. Edit configuration.h and change the value for #define TEMP_SENSOR_0 from 1 to 61.
> 5. In the same file, change the values for #define HEATER_0_MAXTEMP (and all the other heater maxtemp values that come after it) from 300 (or whatever it is) to 315.
> 6. Compile the firmware and flash the bin.
> 
> Before I switched to Klipper, this is how I used the professional firmware without compromising functionality.

@theSt33v, greatly appreciate the steps. Clear and concise, these steps will help others intending to do the same. Cheers!

mm108, 2022-11-17:
> Use the special custom g-code C104 to unlock maximum hotend temperature, for example: `C104 U1 T315`

@mriscoc, thank you so much for the pointers. This will help me tide over the immediate problem that I have - which is to print at 290-300°C. I guess I may need to do something similar to get the bed up to 110°C as the PC filament seems to have issue sticking to the bed. I guess for the bed it would be the BED_MAXTEMP setting in configuration.h. But I guess I can figure that out. Cheers!