mriscoc/Ender3V2S1 #146: Invert Extruder Direction not working? CONFIGURATION_H_VERSION 02010000

Chris-Jayden, 2022-05-07:
It seems that in the latest versions of the uncompiled version, the extruder inversion section doesn't work after compiling and installing into firmware

> // section extruder
> 
> // For direct drive extruder v9 set to true, for geared extruder set to false.
> #define INVERT_E0_DIR true // true for Orbiter
> #define INVERT_E1_DIR false
> #define INVERT_E2_DIR false
> #define INVERT_E3_DIR false
> #define INVERT_E4_DIR false
> #define INVERT_E5_DIR false
> #define INVERT_E6_DIR false
> #define INVERT_E7_DIR false

I have successfully compiled and installed the firmware to my Ender 3 v2 4.2.2 TMC2208, but no matter if I define E0 as true or false, the machine in actual operation has no changes. I am using an Orbiter v2.0, so it needs to be 

> #define INVERT_E0_DIR true // true for Orbiter

rather than 

> #define INVERT_E0_DIR false // false for default extruder

I have found a simple work-around, but it's only a temporary fix -> after installing the firmware, head to the "Advanced" menu then "Filament Settings" then invert the extruder direction via the checkbox.

However, this worries me for the future as I plan to implement 2-in-1 or even 3-in-1 hotend assemblies in the future with multiple extruders on either of an SKR v2.0B F429, Octopus Pro v1.0, or a GTR+M5 v1.0 board with the mriscoc firmware.

Here is my configuration.h  Hopefully you can't spot the error, if any
[Configuration.zip](https://github.com/mriscoc/Ender3V2S1/files/8644521/Configuration.zip)

mriscoc, 2022-05-07:
In the current version DEF_INVERT_E0_DIR is set to false by default, in next version will use the INVERT_E0_DIR from configuration files.

closed 2022-05-08 by mriscoc

Chris-Jayden, 2022-08-05:
As the new version is out now, I was wondering if this feature has been added/modified yet.
Has there been any progress made with DEF_INVERT_E**X**_DIR ?

mriscoc, 2022-08-06:
This firmware supports only one extruder.

Chris-Jayden, 2022-08-07:
Hi Miguel, what steps could/would be needed if, perhaps, I wanted to modify/convert the firmware for use with multiple extruders? Is there really no way for the mriscoc firmware to be (made) adaptable for multiple extruders?

mriscoc, 2022-08-08:
For support multi-extruders, you must re-write the menus and screens where extruder data is shown or edited.