mriscoc/Ender3V2S1 #150: nivelacion automatica no peticion de guardado antes de imprimir

bombardeo38, 2022-05-08:
tengo configurado mi g code de esta manera 

G90 ; use absolute coordinates
M83 ; extruder relative mode
M140 S{first_layer_bed_temperature[0]} ; set final bed temp
M104 S150 ; set temporary nozzle temp to prevent oozing during homing
G4 S10 ; allow partial nozzle warmup
G28 ; Home all axe
G29 ; ABL - BLTouch
G92 E0
M420 S1 Z2 ; USE mesh when calculating z
G1 Z50 F240
G1 X2 Y10 F3000efectua prueba de nivelacion antes de imprimir pero luego pide guardar o cancelar abria posibilidad de que se guardara automaticamente 
otra cosa mire lo que te comente del ventilador si el laminador pone 100% el ventilador se pone en 255 asi que lo solucione bajando el porcentaje

mriscoc, 2022-05-08:
https://github.com/mriscoc/Ender3V2S1/issues/148

Fan 0 - 255 (8 bits) 100% = 255

closed 2022-05-08 by mriscoc

bombardeo38, 2022-05-09:
Buenas podías indicarme donde colocar el c108 m500 que va después del m420

mriscoc, 2022-05-09:
> Buenas podías indicarme donde colocar el c108 m500 que va después del m420

Luego del G29