mriscoc/Ender3V2S1 #151: Bed tramming wizard does not work anymore

RoP09, 2022-05-10:
**Describe the bug**
Bed tramming wizard does not work anymore. Worked fine with previous fw version.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Bed tramming'
2. Click on 'Tramming wizard'

**Expected behavior**
Bed tramming wizard moves to all 4 corners and measures distance with CR/BL Touch

**Actual behavior**
Bed tramming wizard only moves to 1st front left corner and stops after measuring 3 times. 
Gives a positive result (after shorting calculation screen)  showing all 4 edges level even if only 1 was measured.
(all edges showing 0.0 values)

If edges are selected manually only front left and center can be selected. Otherwise it says "**outside of probing area**" if any of the other edges are selected.

**Version:**
**Ender3V2-422BLT-20220419.bin**

mriscoc, 2022-05-10:
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

RoP09, 2022-05-10:
did not help, all physical settings are configured as before still, tried reentering / saving the changes but still the same behaviour:

x bed size: 230
y bed size: 230
x min pos: 0
y min pos: 0
x max pos: 246
y max pos: 231
z max pos: 250

badlyshot75, 2022-05-10:
Probe Margins can be set up to 60 (20 is default), but I see no control item on the menu to change it.

mriscoc, 2022-05-11:
Mesh inset https://github.com/mriscoc/Ender3V2S1/issues/151#issuecomment-1122011429

badlyshot75, 2022-05-11:
OK Thanks. It just was not intuitive for us slow folk.

mriscoc, 2022-05-11:
> OK Thanks. It just was not intuitive for us slow folk.

Yes, something to correct.

eblieb, 2022-05-17:
Also having this issue