mriscoc/Ender3V2S1 #152: G12 Support

TimDowker, 2022-05-11:
Would it be possible to enable the Marlin G12 command in this firmware?

https://marlinfw.org/docs/gcode/G012.html

At first it may just be easiest to enable to command and people can use a terminal to issue it or include it in their generated gcode.

A user interface could probably be done afterwards.

mriscoc, 2022-05-13:
That could be planned for a future release, but currently it is difficult to add features because the RCT SoC's are close to the limits of flash memory. In any case, G12 can be emulated using simple G0 movements.

Chris-Jayden, 2022-05-16:
> That could be planned for a future release, but currently it is difficult to add features because the RCT SoC's are close to the limits of flash memory. In any case, G12 can be emulated using simple G0 movements.

Would an SKR v2.0B STM32F429VGT6 board have enough memory? Rather, would the aforementioned board even work with your firmware? I'm looking to replace my stock motherboard with the SKR v2.0B STM32F429VGT6 so that I can make a dual-extrusion Ender 3 v2; hopefully, it's compatible with this firmware and its variants

mriscoc, 2022-05-28:
> Would an SKR v2.0B STM32F429VGT6 board have enough memory? Rather, would the aforementioned board even work with your firmware? I'm looking to replace my stock motherboard with the SKR v2.0B [...]

Now the firmware has BTT SKR Mini E3 V3 support, maybe you can use the configuration files as foundation to build an environment compatible with your board.

Chris-Jayden, 2022-07-16:
> Now the firmware has BTT SKR Mini E3 V3 support, maybe you can use the configuration files as foundation to build an environment compatible with your board.

I tried running a mrsicoc compile today for the SKR v2.0B-F429 VTG6, but I'm running into a categorically restrictive error.

SKR 2.0 boards don't support "PRINTCOUNTER", so it has to be disabled to successfully compile on both standard and Jyer's Marlin.

For mriscoc Marlin, however, ProUI requires "PRINTCOUNTER" to be enabled
[image]

I'm at a loss as I'm unable to proceed forward. Let me know your thoughts.

mriscoc, 2022-07-16:
Please wait until the release of the new sources, PRINTCOUNTER and other features will be optional.

closed 2022-10-04 by mriscoc