mriscoc/Ender3V2S1 #154: Auto build mesh gives different results pre and post power cycle

jchu1988, 2022-05-12:
**Describe the bug**
Mesh collected is different depending on how long the printer has been powered. In subsequent Auto Build mesh, the earlier probe points, the higher the Z value. on a 5 x 5 probe, starts at 0.03 and descends to -0.75 mm.

**To Reproduce**
Steps to reproduce the behavior:
1. Preheat bed
2. Auto build mesh and tram bed until with 0.1 across all points
3. wait 24 hours
4. preheat bed
5. Auto build mesh
6. Mesh will show the earlier probe points to be higher and descend as more points are probed
7. Power cycle
8. preheat bed
9. Auto build mesh

**Expected behavior**
consistency in values when powered for prolonged duration

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version (please complete the following information):**
Firmware version and release date
2.1.0.1 MRIscoC

**Additional context**
Also found the M48 Probe test reported significantly higher value (0.002 typical, vs 0.06).

jchu1988, 2022-05-13:
After 4 hours on heated standby, printer Auto Build Mesh started to drift. By using M48, can see the value increased to 0.008. Disabling HS mode, reduced the value back to 0.002 and bed mesh was within +/-0.04.

mriscoc, 2022-05-13:
Not only is the bed heated after many hours, but through convection also the other pieces of the printer, so it is expected to have some deviations. This is a sub $500 printers then we cannot expect to high accuracy. Marlin firmware try to achieve the best measurements.

jchu1988, 2022-05-13:
Then please explain why turning off HS mode resolves or a quick power cycling resolves the drift? 

Thermal is a slow process in comparison.

mriscoc, 2022-05-13:
Turning off the HS mode improves the repeatability and accuracy of some leveling sensors, especially the CRTouch.

GootyS, 2022-05-14:
Where can i find the HS mode? Im finding the Auto build mesh not working that great.

jchu1988, 2022-05-14:
> Where can i find the HS mode? Im finding the Auto build mesh not working that great.

Advanced > Probe settings > Enable HS mode
uncheck.

jvcjbl, 2022-05-18:
I was seeing the same results. It ended up being my bushings. I switched back to the yellow springs. Trammed to 0.0 on all the corners. Ran 9x9 UBL mesh using CR Touch and HS on and got within .5 across the whole bed. Let the printer sit turned off for 3 days and when I turned it back on ran the tramming wizard and everything was still 0.0 so I decided to do the mesh creation again. It only diverted .1 from what I saw. Maybe a freak occurrence but I was constantly having bed tilting issues with the bushings. Just my $.02, YMMV

Victordiazbis, 2022-05-19:
I have the opposite issue. The bed measuring is quite repetitive, within a narrow margin, but it shows strange results, with peaks and valleys across X axis. Test prints demonstrate that those undulations are not real. The easy explanation is an issue with printer kinematics, but ti my eyes it looks ok. For the moment I am using the auto mesh as starting point, and the I edit the mesh point by point after many test prints. It is not a matter of microns, I am talking about tenths. In any case, after this terrible process, the printer performs well and it is possible to get a decent and repetitive first layer. Any idea of what's happening?

Victordiazbis, 2022-05-21:
I have tried with both versions,  UBL and standard leveling. Same results. The leveling is more or less consistent across y axis, but is a complete disaster across X axis. It reflects also in bed tramming. A perfect automatically trammed bed, is clearly tilted to the right side. Every mesh that I generate, no matter the feed rate, or the HS activation/deactivation are full of vertical stripes alternating positive and negative values with no sense at all. Could it be a defective CR touch? All axis goes smoothly and everything looks well aligned. Any help is really appreciated

mriscoc, 2022-05-23:
Full vertical stripes can be an indication of dents in eccentric wheels of X axis.

Victordiazbis, 2022-05-23:
Very intelligent observation, but the printer is new. I thought about a foreign object in X axis, but it prints gorgeous, so I tend to think I Z axis. Could for some reason the same action of the steppers produce diferente Z motion for different positions of x axis while doing the bed measuring? I checked the excentric oz z axis, and it looks ok. Z axis goes quite smoothly.

Victordiazbis, 2022-05-24:
Miguel, eres el puto amo. Mil gracias. Efectivamente una de las ruedas estaba un poco ovalada. Ahora falta hacer el ajuste fino, pero al menos la medición de la cama tiene sentido 
[image]

closed 2022-06-08 by mriscoc