mriscoc/Ender3V2S1 #158: Mesh data is flipped

seth199, 2022-05-17:
**Describe the bug**
"View mesh" does not match mesh shown when M420 V1 command is issued. The result is flipped about the horizontal axis. I do not know if this flips the mesh when printing though it seems it does.

**To Reproduce**
Steps to reproduce the behavior:
1. create a mesh and save
2. Advanced > Mesh Leveling > View Mesh
3. Using Pronterface issue M420 V1 command

**Expected behavior**
Ignoring the rounding of numbers the mesh data should be identical between the LCD display and the Pronterface console.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

[image]

**Version (please complete the following information):**
2.1.0.1 MRiscoC (non UBL)
Apr 20 2022 00:01:07

**Additional context**
Add any other context about the problem here.

mriscoc, 2022-05-17:
It is not flipped; the text table shows the 0,0 in the upper left corner, but physically the 0,0 is in the front left corner.

closed 2022-05-17 by mriscoc

seth199, 2022-05-17:
Thanks for the quick reply. Sorry for the dumb question.