mriscoc/Ender3V2S1 #162: Wierd noises from Motherboard(i think) when trying to use preheat PLA

BuzzBill, 2022-05-19:
**Describe the bug**
i just tried to preheat the printer using the preheat  PLA and immediately wierd noises. Everything else works fine.

**Version  Latest Version

BuzzBill, 2022-05-20:
**Update  
i opened the board case and the fan is making the noise every 2-3 seconds a sound like a wild spin...
The printer is brand new and made no noise until today

mriscoc, 2022-05-20:
The board fan is connected in parallel with the part cooling fan, you can test it in Control / Temperature / Fan speed.
[image]
Some fans start to make sounds after some time of use or because dust accumulation.
It is not a firmware issue.

closed 2022-05-20 by mriscoc