mriscoc/Ender3V2S1 #163: Bed Tramming and Create Mesh tools do not use Physical Bed Settings

enderextender, 2022-05-20:
**Describe the bug**
When adjusting the Physical Printer Settings (max X, max Y, max Z) and setting bed size, these settings are not utilized when performing Bed Tramming Wizard or Mesh Bed Levelling, instead, the hard coded Ender 3 size (220x220) is used.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Advanced, Physical Settings, set X size, Y Size (I am using 400x400 on an Ender Extender).  Set Max X and Max Y to 410/410
2. Store Settings
3. Go to Bed Tramming Wizard
4. Observe Bed Tramming Wizard only using 220x200

**Expected behavior**
Bed Tramming Wizard and Mesh Levelling should respect adjusted physical settings.

**Version (please complete the following information):**
20220419

mriscoc, 2022-05-20:
Please complete the version of your firmware, ABL or UBL. Did you set up your mesh inset? 
Use Advanced / Mesh leveling / Mesh Insets / Maximize area (and optional center area)

enderextender, 2022-05-20:
I used UBL.  I will check the values you specified.

mriscoc, 2022-05-28:
No more feedback?

closed 2022-05-28 by mriscoc

enderextender, 2022-05-31:
No, it works, it’s more confusing than it needs to be but it works.