mriscoc/Ender3V2S1 #168: Tune menu when printing from Octoprint

vaskito85, 2022-05-22:
**Describe the bug**
When I print from Octoprint I can´t have the tune options, pause or cancel button as when i print from the SD card. This is normal or im doing something wrong?

**Screenshots**

[image]
Printing From the SD card

[image]
Printing from octoprint

**Version (please complete the following information):**
Latest version April 20 for Ender 3V2 4.2.2 board

mriscoc, 2022-05-22:
Please read de Wiki https://github.com/mriscoc/Ender3V2S1/wiki/Octoprint

closed 2022-05-22 by mriscoc

vaskito85, 2022-05-22:
Miguel, Acabo de darme cuenta de que los dos somos compatriotas latinoamericanos. 
Habia hecho todos los pasos de la wiki y seguia sin aparecer. 

Re-instale el firmware y re-instale el octoprint y ahora si funciona. Perdon por generar el "Issue".

Abrazo desde Argentina!