mriscoc/Ender3V2S1 #169: Printing does not start

Slavikpd, 2022-05-23:
Hello.
I have 3D printer Ender  3 V2,with mainboards 4.2.2 and with Chip STM32F103. I download the firmware Ender3V2-422BLT-UBL-20220419.bin. To settings of slicer I (PrusaSlicer 2.4.2+x64 and CrealitySlicer 4.8.2-96) put G-Code´s: 
M190 S70
G28
G29 P1
G29 P3
G29 F10
G29 S1
G29 A
G29 L1
M500

After start of the printing calibration will take place and printing won´ t start (all motors just stand), but on the display it show process of printing very fast. I´m attaching short video. When I repeat teh printing it will start normaly. Some times its start on the second or third try. It happens even when I didn´t put G-Code´s to the slicer and I do calibration manualy. Can you please help me?
Thank you.

https://user-images.githubusercontent.com/99100274/169777943-4bea41a2-68e1-47f4-b01c-98ff06f02164.mp4

[Cube 10x10x10mm G-Code.txt](https://github.com/mriscoc/Ender3V2S1/files/8752832/Cube.10x10x10mm.G-Code.txt)

mriscoc, 2022-05-23:
Re-create the printer profile or try with another slicer to test.

mriscoc, 2022-05-28:
No more feedback?

closed 2022-05-28 by mriscoc

Slavikpd, 2022-05-28:
I'm sorry, but I didn't have time to deal with it. I've tried setting up a profile, another slicer, but it still does. The strange thing is that it does it irregularly and on different files stl. I'll search and see what can be done. Thanks for help.