mriscoc/Ender3V2S1 #174: when i use 'How to generate a gcode preview' and export the gcode 【PrusaSlicer encountered an error】

LiFulian, 2022-05-29:
when i use 'How to generate a gcode preview' and export the gcode
[image]

【PrusaSlicer encountered an error】
Post-processing script "C:\Python310\python.exe" "C:\Users\lifulian\AppData\Roaming\PrusaSlicer\scripts\CreateHeaderAndThumbnail.py" on file C:\Users\lifulian\AppData\Local\Temp\.17656.gcode.pp failed.
Error code: 1

CaVO32, 2022-05-29:
i also have the same identical problem

mriscoc, 2022-05-29:
Current python script has support up to Prusaslicer 2.4.0, the script will be updated soon.

mriscoc, 2022-05-30:
Updated super/Prusa Slicer thumbnail post processing script, please re-download it.

closed 2022-05-30 by mriscoc