mriscoc/Ender3V2S1 #175: M420 S1 does not enable mesh

DevlshOne, 2022-05-29:
**Describe the bug**
In the Pro Firmware for the F4 STM32F4 chip, sending M420 S1 in the start GCODE does not enable bed mesh.  This GCODE works correctly on the STM32F1 firmware.

**To Reproduce**
Steps to reproduce the behavior:
1. Use Auto Build Mesh to build a mesh
2. Click Save, then Continue
3. Go to terminal and enter M420 S1. Mesh will not enable.  Line under Z does not turn red.
4. No error reported, just not enabling mesh

**Expected behavior**
Mesh leveling should be enabled and the line under the Z icon should change from white to red.

**Screenshots**

**Version (please complete the following information):**
Ender3S1-F4-20220527.bin

DevlshOne, 2022-05-29:
I'm sorry, I don't code in Python and don't use VSCode so I'll have to speculate here.  I think the issue may be that `RESTORE_LEVELING_AFTER_G28` is not enabled in the build and I am issuing a G28 in my start GCODE. I'm going to try with `G29 A` followed by `M420 S1`

mriscoc, 2022-05-29:
For the F4 there is not a UBL version, then the correct way to enable the mesh leveling is with a M420 S1 after the G28 (home axes). The white lines colorize (white to blue or red depending on the icon set).
The mesh leveling is only activated if there is a **VALID** mesh leveling in RAM. There are two ways to have a valid mesh in ram, after a successful Auto leveling or loading a valid mesh leveling from EEPROM.

DevlshOne, 2022-05-30:
> For the F4 there is not a UBL version, then the correct way to enable the mesh leveling is with a `M420 S1` after the `G28` (home axes). The white lines colorize (white to blue or red depending on the icon set). The mesh leveling is only enabled if there is a **VALID** mesh leveling in RAM. There are two ways to have a valid mesh in RAM, after a successful Auto leveling or loading a valid mesh leveling from EEPROM.

I'm going to make my first attempt at compiling a UBL-enabled version. Thanks.

closed 2022-05-30 by DevlshOne

Friday13th87, 2022-11-03:
Is there any difference in using UBL with start gcode
```
G29 A
G29 F2.0
G29L0
G29 J1
```

instead of just
`M420 S1 Z2`

I was normally using the G29 Codes to activate and use UBL?
Do i need to activate the Mesh in the advanced settings? It is disabling itself as far as i can see.

If i have a mesh saved in slot 0 and using cura - what start g-code would be recommended for using the mesh?
Thanks in advance

mriscoc, 2022-11-04:
```gcode
G29 L0 ; load mesh from slot 0
G29 A  ; Active UBL
```